portapack-mayhem/mayhem-firmware #373: DAB receive/transmit??

sebastianheyn, 2021-06-08:
**Is your feature request related to a problem? Please describe.**
Hi, I'm using the hackrf for DAB transmission. mostly captured .eti files. however, i cannot make use of the portapack...

**Describe the solution you'd like**
An integration of ODR-DABMOD would be excellent. the IQ data would be created on the fly ?? 

at the same time, reception and capture of regional DAB stations would be an excellent addition to the existing features of the mayhem firmware.

fcqpl, 2021-06-12:
Popably it won't work on Portapack. Audio transcoder takes too much CPU.

sebastianheyn, 2021-06-12:
Hey guys, when using a eti file, the audio is already transcoded.  So eti is basically a format that contains all the single streams of a station. If you receive a dab frequency, and it's demodulated, you can store to .eti directly. I can send example files if you want to play around with it. Cheers,BastuVon meinem/meiner Galaxy gesendet

anigwei, 2022-02-19:
Hi,

I'm a future Portapack user and DAB+ player would be a very interesting feature for portable DX.

For example, welle.io software is ported to ARM (Android) and in fact it works with lowend phones... Would be an option to "port" it to Mayhem?

As @sebastianheyn says, broadcasting would be also possible by entering a .eti file that you could prepare previously in a computer. So the file only has to be transferred to hackrf (in a Linux computer is done with "hackrf_transfer" command). No CPU processing.

MerApple, 2023-10-17:
You could connect portapack to a raspberry pi to get more CPU power and then use for an example SDR angel to listen to DAB. This could still be a portabel solution with battery possibilities