portapack-mayhem/mayhem-firmware #375: SSTV TX app cannot read files smaller or larger than  240kb

JoshuaMorris, 2021-06-10:
**Describe the bug**
Attempting to use any image other than the sstvtest.bmp or 320356C.bmp images included in the sdcard files fails to be read by the sstv transmission app. the included 320240BW.bmp file and others are unavailable to transmit. only perceived difference is the file size of the images 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Tools'
2. Go to 'Utilities'
3. Go to 'File Manager'
4. Open /SSTV folder
5. Confirm multiple files exist of various files sizes 
6. Exit to main menu
7. Go to 'Transmit'
8. Go to 'SSTV'
9. In file selection, rotate dial. You will only be able to access files that are 240kb in size to transmit.   

**Expected behavior**
You should be able to select any bmp file in the SSTV folder

**Affected versions**
* Latest Stable Release

ArjanOnwezen, 2021-06-11:
Only 24 bit BMP files with 256x320 pixels are supported. It's not a bug, it's a feature...
So please  convert your BMPs to this format. I guess we should change the remaining file(s) in SSTV folder in this repo to avoid confusion, unless someone changes this.

JoshuaMorris, 2021-06-11:
Is this a hardware limitation?

ArjanOnwezen, 2021-06-11:
It looks like a software limit.

JoshuaMorris, 2021-06-11:
Okay, It would be great to be able to not have this limitation or feature in the code. @ArjanOnwezen Thank you for looking into it. I should have dug into the code more myself, still working to get acquainted with the codebase