portapack-mayhem/mayhem-firmware #377: Problem upgrading the firmware

MHNikkhah, 2021-06-11:
hi
I saw the video and try to upgrade the firmware step by step but unfortunately after upgrading from windows the hackrf device was having problems. All LEDs turned  on and LCD dos not show any things and the computer USB dos not recognized hackrf any more. I had to update the device via DFU mode and now the previous portapack firmware version (havoc) dos not exist any more and i can just connect hackrf to computer.

eried, 2021-06-11:
Check the video about the DFU process carefully again. It explains how to recover from that. Try different USB cables too.

MHNikkhah, 2021-06-12:
THANK YOU
I tried updating with another computer and another USB cable. 
it is working so good

closed 2021-06-12 by eried