portapack-mayhem/mayhem-firmware #391: My PortaPack H2 has a speaker that doesn't emit audio; headphone jack works

emory, 2021-06-30:
**Describe the bug**
I can't hear anything from my PortaPack H2.

**To Reproduce**
Steps to reproduce the behavior:
1. Use any of the features that listen to live audio 
2. Nothing audible unless headphones are plugged in

**Expected behavior**
I expected the speaker to play audio. I wrote the person I bought from and was told: 

> "Yes, you need to enable it and turn up the volume,thanks.​‌‌‌‍‍‍‌‌‌‍‌‌‍‍‌‍‍‍‍‌‍‌‍‍‍‍‌‌‌‍‌‌‍‌‌‍‍‌‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‍‌"

I don't know how to enable it in Mayhem. I saw the option to hide the speaker icon in the menubar for non-H1 platform hardware but that's it.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* 1.4.0 is what I'm running on the device and that is what was shipped to me. I've been writing my own scanlists for where I spend most of my time but have not yet built or installed firmware on the device yet.

I don't know if this is a Mayhem issue or a PortaPack 3rd Party Weirdo issue. I've been unable to find any software-side answers for this behavior, I lost a µSD card in my PortaPack so I had to open it up and I saw that the speaker is hitched up but I didn't get a meter or test anything and now I feel a little embarrassed about that but I don't know what the expected behavior would be to validate it's connected properly with the wiring exposed to me.

So far I'm really enjoying Mayhem. I need to learn more about the modules and how it all works, I haven't figured out how to save discovered devices in the Bluetooth sweeper or apply friendly metadata or descriptions of things on-device but I'm having fun learning. Hooked it up via USB to a macOS workstation last night and it worked as expected with

ArjanOnwezen, 2021-06-30:
Is speaker enabled (green icon top bar)? If so, it's most likely a hardware issue.

teixeluis, 2021-06-30:
@emory is there a chance speaker contacts could have come into contact with the chassis or anything on one of the boards? You may want to verify if the audio amplifier chip (in some boards its a small 8-pin chip) is ok. Try to check if it is blown. If that is the case, it may have a mark or something.

These amplifiers are quite sensitive to shorts to the chassis GND, as these have floating outputs (bridged amplifier).

emory, 2021-07-15:
that may well have been in, @teixeluis — now it's been working but all i did was open 'er up to confirm. the icon shows up now (i have the option to disable the icon for the H1 spec portapacks active i.e. disabled the icon that is the wrong speaker) and i can get audio. i'm closing and if it happens again i'll try to confirm a physical root cause and update here.

closed 2021-07-15 by emory