portapack-mayhem/mayhem-firmware #392: Bricked after attempting new firmware update

LuvHacOne, 2021-07-08:
Plugged in to PC - detected on USB, firmware appears to be loading, terminates on windows with an error message, GAME OVER.
When power applied, 3v3 is bright, TX is about 50% LED brightness, USB and RX are both at about 5% brightness. No access to DFU mode, just an expensive brick.

gadjodilo83, 2021-07-09:
remove the portapack and try again...

LuvHacOne, 2021-07-09:
Tried that already, Hack one is no longer recognised on any usb port on any of my computers.

janicektrnecka, 2021-07-09:
Ended up with the same issue, did this to recover - reset to DFU, flash dfu just to hackRF, do not reset or unplug and flash mayhem. I had to split the portapack and hack rf if I remember it correctly.  The behavior was something like this > device bricked, lcd wont light up, just 3,3V led and faint RX TX leds , no 1,8V present. Reset to DFU, flash it and If I didnt reboot or power cycled it it registered as usb hackrf and was working under eg SDRsharp. after reboot it was bricked again. What helped me, was flashing it with portapack just after DFU hackrf flash. I was doing it in 4AM , so this might be a bit hazy...

eried, 2021-07-12:
As @janicektrnecka you should do the DFU procedure

closed 2021-07-12 by eried

LuvHacOne, 2021-07-13:
running the dfu gives me the end result of
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

n3o59hf, 2022-07-31:
Had the same issue, was able to recover by using shorter USB cable (same method described before - DFU flash, then without reset or reconnect Mayem flash). "Unable to read DFU status after completion" seems to be normal and does not cause problems.

DadMych, 2022-08-31:
Shorten USB cable worked!

comradecowboy9, 2023-04-29:
How are you all flashing mayhem after the hackrf FW when they're separated? I can get the dfu load to work and the USB light to come on but I'm not sure how to flash mayhem after that