portapack-mayhem/mayhem-firmware #397: Scanner App does not  sweep through complete manual interval

maichai, 2021-07-28:
**Describe the bug**

When manually setting a scanner intervall and step, the scanner app does not sweep through the complete interval from Start to End, unless the Step parameter is large.

**To Reproduce**
Steps to reproduce the behavior:
1. set Start to e.g. 400
2. set End to e.g. 470
3. set Mode to WFM
4. set Step to 50kHZ
5. click Scan
6. watch the scanner sweep from 400 to ca 430 and then start over at 400. It does not sweep through to 470.

**Expected behavior**

The scanner shoudl sweep the whole frequency interval, no matter what Step has been defined.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* v1.4.0

GullCode, 2021-07-28:
That's because the scanner app is using a list of precalculated FREQS and
not a start-end-increment.

The search app in a pending PR is doing what you're asking.

maichai, 2021-08-01:
That is still a bug in the scanner, right? Why would be be able to set start and end if the scanner would not use these values to scan that range?

GullCode, 2021-08-01:
This is how the scanner app was designed, we can not say it's a bug. You
can try the search app I  mentioned.  It's doing what you want.

hitblackjack, 2022-02-05:
v1.4.0    
the scanner app has two modes:
1) load frequency list file, it begins automatically;
2) give the start, end, and step, it generates frequency list on the fly, then click mannual(scan) to begin.

both mode have a DB MAX=500 limit. 
so the app just scans between the [start frequency] and [the 500th frequency],
all the frequencies beyond the 500th will be ignored!
btw, the scan rate is about 20 channels per second, not very fast.

hitblackjack, 2022-02-05:
> That is still a bug in the scanner, right? Why would be be able to set start and end if the scanner would not use these values to scan that range?

v1.4.0
the scanner app has two modes:

load frequency list file, it begins automatically;
give the start, end, and step, it generates frequency list on the fly, then click mannual(scan) to begin.
both mode have a DB MAX=500 limit.
so the app just scans between the [start frequency] and [the 500th frequency],
all the frequencies beyond the 500th will be ignored!
btw, the scan rate is about 20 channels per second, not very fast.

euquiq, 2022-02-05:
About the speed, it is "as good as it gets" since there is a 50ms period of "re-stabilizing" when changing the freq over each step.

GullCode, 2022-04-13:
This can be closed as it's not a bug but the way the scanner is working.
Feel free to fork my search app or wait for it to be merged :-)

closed 2022-04-13 by jLynx