portapack-mayhem/mayhem-firmware #400: new app: LoRa receiver

Gurkengewuerz, 2021-07-31:
i would really love to see a LoRa decoder in Mayhem. It's really a very cool protocol and implementing a decoder is a huge step for Mayhem.

In my opinion a view for receiving LoRa signals and decode the header data like preamble or mac as hex and the payload as hex/char is enough and a neat start.

I guess implementing Chirp Spread Spectrum modulation is the biggest issue. Perhaps one or the other developer already has experience.

What are your opinions?

Gurkengewuerz, 2021-07-31:
For the record: I can help with this but i have really a problem of understanding the CSS modulation and implementing it. Hardware development is my business.
If someone is interested i can help with the views or decoding the raw data.

ssinger99, 2021-09-21:
This would be great ... just getting into the HackRF and LoRAWan stuff at 915mhz!

strijar, 2021-11-06:
You can take the module for gnuradio as a basis. For example - https://github.com/rpp0/gr-lora

smannchen, 2022-04-02:
This needs to be a thing lol. Or even better to be able to use the portapack as a meshtastic device.

andrewsclapp, 2024-01-01:
Also +1.  My toy collection contains enough of the right things to test code if someone can develop it.  I have a portapack and some heltec radios that I can set up a point to point test with.

Miahporter, 2024-02-06:
Would love to see a Meshtastic app on the H2. +1  for Meshtastic app integration.

Radiomix2000, 2024-02-14:
> You can take the module for gnuradio as a basis. For example - https://github.com/rpp0/gr-lora

that seems to not full implementation.
https://github.com/tapparelj/gr-lora_sdr here is repo, completed and fully compatible with known LoRa chips

demondragonwren, 2024-02-22:
I'm very excited to see this implemented.

VigibotDev, 2024-05-10:
+1 LoRa stuff (i like meshtastic)

Hime0698, 2024-08-22:
here to pitch in excitement for LoRa especially meshtastic support

eried, 2024-08-22:
+1 this would be nice

strijar, 2024-08-30:
I found a very compact and lightweight LoRa decoder in C. Maybe I'll start working on this.

VigibotDev, 2024-08-30:
It's awesome because LoRa is a lot of work to decode in digital signal processing vs. basic GFSK or any others...

htotoo, 2024-08-30:
> I found a very compact and lightweight LoRa decoder in C. Maybe I'll start working on this.

Could you link it please?

173duprot, 2024-08-30:
I would love to see the reticulum network added. It fits this project really fuckin well. 
https://reticulum.network/

strijar, 2024-09-02:
https://dlbeer.co.nz/articles/lora/lora.c

nbirdmann, 2024-11-04:
https://github.com/htotoo/ESP32-Portapack/
Maby we can add it into this

gullradriel, 2024-12-04:
For the records, the issue was also duplicated in https://github.com/portapack-mayhem/mayhem-firmware/issues/693 and https://github.com/portapack-mayhem/mayhem-firmware/issues/2171. These issues were closed so we are only keeping track with this one.

SakNikStudio, 2024-12-18:
High hopes on this one, the idea is just perfect.

Waycco, 2025-01-19:
Could it be possible to run SX1262 chip as a small add on plug in device in GPIO port and use that for LoRa comms processing? Just an idea....

htotoo, 2025-01-20:
> Could it be possible to run SX1262 chip as a small add on plug in device in GPIO port and use that for LoRa comms processing? Just an idea....

only if you add an esp (or similar) to it, since iirc sx is spi, not i2c.

Arkwin, 2025-04-16:
+1 on this too.

htotoo, 2025-07-09:
The module for esp32pp is work in progress:
https://github.com/htotoo/EspMeshtasticCompact

Now it can receive and send unencrypted messages to nodes and to default encryption primarily channel.
Supports some telemetry rx, ...

But before i integrate this, you should suggest some real usecases. Because an app for portapack, where you can send / rx messages are not so useful, since meshtastic has a great mobile app, and using that is much more easy.
So pls provide real use cases.

ckuethe, 2025-07-09:
Meshtastic is one app that can run over LoRa. People may be interested in other systems that also use LoRa as a transport... There's also dump1090 but mayhem can still be used to receive ADS-B.

htotoo, 2025-07-09:
don't mix other protocols here, focus on lora.

the "it can receive lora messages" is still not really an use case. you can go to waterfall, and it receives the lora, but not processing it. and it covers that, "it receives it" :)

But just watching the raw hex packets arrive, may not be the useful stuff i want to create.
so for example with meshtastic, i can imagine that you could remote manage (without screen!) the portapack via USB shell commands.

But any good idea, that would be a real use case would be good.

ckuethe, 2025-07-09:
I drew a parallel to adsb because there are prettier decoders like dump1090, or radiosonde_auto_rx for weather balloons, or a good commercial receiver for broadcast FM, but mayhem makes for a decent view to all of these and more.

pagers, ais, adsb, etc. can be logged to sdcard for later processing with other tools, I don't see what's wrong with logging or printing raw LoRa packets.

nbirdmann, 2025-07-09:
Being able to find node locations by signal strength would be awesome

ShakataGaNai, 2025-07-09:
As Meshtastic has a mobile device/app that can show you messages, put dots on a map, etc. I'd say that "lower level" logging and display in the real usecase. Being able to see (and save) packets in a more raw form (json level) gives you access to a lot of the data that is otherwise "hidden" in the app. More important recording other data like SNR gives you an option to foxhunt nodes. To that end, being able to filter by sender, strength, hop or packet type would also be valuable. 

It doesn't even really need the ability to send messages, as the existing meshtastic devices can do that. But say pair a standard node that's requesting updates from your target (or sending them DM's, which their device auto-acks), with a portapack using a small directional antenna... means you could potentially fairly quickly triangulate a nodes location.

Sazoji, 2025-07-10:
> But just watching the raw hex packets arrive, may not be the useful stuff i want to create. so for example with meshtastic, i can imagine that you could remote manage (without screen!) the portapack via USB shell commands.

See, this is what I was interested in with the remote port dev, you can only cover one part of the spectrum, but you can use the portapack as a trigger or another device as an input. Making a repeater or a remote setup with two radios only communicating via serial limits you to digital modes, but it's already possible (I was looking at using esp32 VHF band radio to transmit pre-recorded signals off of its SD card while the portapack is receiving signals at a different frequency, a lora radio would be able to handle coherent data tho).

Something that utilizes the waterfall and reports back via lora could be possible with just the portapack, have it report on traffic conditions in a different by sending back the accumulated FFT periodically. I think the single antenna options are pretty limited for transmitting on or interacting with lora networks tho. Better to have an independent lora device interact with the portapack, so its antenna is fully open for doing its job remotely.

A mesh of portapacks returning remotely set SDR options via a lora network would be an interesting team method to map out a signal, but that's moving ALL the Lora tx/rx work to a separate device, not really part of this PR.

melroy89, 2025-10-03:
> in short: no the device is not capable of this computing by itself.

[Meshtastic_SDR](https://gitlab.com/crankylinuxuser/meshtastic_sdr) (running on any Linux machine) can be perfectly used together with the HackRF device. Not sure what you are talking about exactly.

You mean mayhem firmware / HackRF device will never be able to create a local app because the compute is not available on the device itself?

htotoo, 2025-10-04:
If you use it as hackrf, you can use it.

W1BTR, 2025-12-16:
Bumping this, was surprised it's not an app. Would be great to support Meshtastic / Meshcore!