portapack-mayhem/mayhem-firmware #401: in scanner app audio distorted and incomprehensible AM and NFM

lazlooose, 2021-08-01:
Hello, while using the scanner app running Mayhem 1.4.0 and 1.3.0 the audio is distorted and inaudible if I switch to the audio receive mode it works fine but the audio in the scanner app after it locks to a station is horribly distorted and quiet. this is true of AM (DSB - CB band) and NFM (FRS Band). has anyone else had this problem?

Steps to reproduce the behavior:
1. Go to scanner app
2. find an audio signal on AM or FM (I used the CB band And FRS Band)
3. After station locks audio is incomprehensible and distorted 
4. however if you click on the audio button to leave the scanner and go to the traditional receive audio app the audio is clear

**Expected behavior**
I expected to hear audio clear, same as in audio receive app.

**Affected versions**
* Latest Stable Release 1.4.0
* and 1.3.0

euquiq, 2021-08-09:
Is this a consistent problem ? I.e. it happens every time the scanner locks into a -say- FM station ?

lazlooose, 2021-08-15:
yes, it is, everytime and with all modulation types too, FM and AM too

euquiq, 2021-08-15:
Did you try an earlier version,just before 1.3 of the firmware ? I have no distortion at all ! I am using my own firmware version, which would include a mix of features (up to 1.3) but on the scanner side (and MIC TX /RX) would be pre 1.3.

euquiq, 2021-10-03:
any news about this ? any other user evidencing this problem ?

ArjanOnwezen, 2021-11-11:
Closing issue, since it's unreproducible. @lazlooose  feel free to open issue with more info. Looks like a faulty device

closed 2021-11-11 by ArjanOnwezen