portapack-mayhem/mayhem-firmware #402: Speaker Control WM8731S Version

Jensen567, 2021-08-04:
[image]

I have this version of the Portapack H2 (+?), I have installed a small speaker into the enclosure, but I can't figure out how to disable it.  I don't see a speaker icon anywhere, and the speaker continues to output when headphones are plugged in.  It is supposedly version 1.4 of the software.

Is this a limitation of the WM8731 versions, or am I missing something?  If it is a limitation, I think I will just buy an AK4951 and swap the chips, as my board seems to have a footprint available for both.

Sorry if this is not the correct place to post this.  It is also not keeping time after power off, but I need to check to make sure the coin cell isn't dead before I call that an issue.

eried, 2021-08-06:
Why you want to disable the speaker? The speaker icon is for switching to the headphones in the H1

Jensen567, 2021-08-06:
Currently the speaker keeps playing when i have headphones plugged in, I would just like to have one at a time.  Would not bother me if headphone plug stayed active all the time, I can just unplug those, but I would like to be able disable the speaker to stay quiet late at night.

jumbo77, 2021-08-08:
SPEAK Mute Function is only for AK4951,it have Internal AMP for SPK,WM8731 dont have AMP so i use NS8002 amp for SPK come form Phone.

Jensen567, 2021-08-08:
I actually found an option under interface settings to put the speaker icon back into the top bar, and this does work to disable the speaker.  Thanks for the replies.

closed 2021-08-08 by Jensen567

hallard, 2022-03-04:
Hi,
Could you tell us where did you connected your speaker, in connector on SPK1 and SPK2?
I'm asking because I have the same board and unable to get speaker to work wiring with the connector, tried 3 and no luck (of course select speaker in menu bar it's green). Plug an headphone I got the sound it's no problem

Looks like traces are going to missing components, so I'm just asking how dod you suceed to have sound on speaker

[image]

eried, 2022-03-07:
I think that component is usually not populated, check the connections here: https://github.com/eried/portapack-mayhem/wiki/Internal-speaker

hallard, 2022-03-07:
Yeah looks like all boards are like that, but trust me I'm not a noob but on my board this is not working, I'll give a try to see output signal with a scope to be sure if you confirm that to works speaker icon on the menu bar just need to be gree?
Thanks

eried, 2022-03-07:
Do you have an H1 or H2? for the H2 you can relay on the automatic switch between headphones and speaker. In settings / audio disable the speaker thing

eried, 2022-03-07:
(as explained in the link I shared before, specially -> https://github.com/eried/portapack-mayhem/wiki/Internal-speaker#h1-specific-setup)

hallard, 2022-03-07:
Got an H2+ exactly the same (same version printed on PCB) as the 1st picture in this issue from @Jensen567 except that the headphone header is better solderer :-)
Can't believe the 3 speakers I tested are failing