portapack-mayhem/mayhem-firmware #403: Regional/national maps

SMILLEE2, 2021-08-10:
Hi I recently bought a portapack h2. But when using the see on map function in the AIS boats section I get a world map, is it possible to zoom in or get a national map or even a regional map?

Thanks in advance

eried, 2021-08-10:
There was someone thinking on revamping the map system again to use tiles (the current one is limited) but no progress has been made. In the current state, it is impossible to zoom, but you could theoretically [create](https://github.com/eried/portapack-mayhem/wiki/World-map-generation) a bigger map file... I think it is almost maxed out though

SMILLEE2, 2021-08-10:
A oké thanks for the quick response. I will just wait it out for the new feature to be released some time🥲

closed 2021-08-10 by SMILLEE2

eried, 2021-08-11:
There is an option of creating a specific issue with a request and adding a bounty to it https://www.bountysource.com/teams/portapack-mayhem 😎 in case you really dream for that feature 🤣