portapack-mayhem/mayhem-firmware #404: Kitematic Docker will not compile completely.

stephen3303, 2021-08-10:
I am following the steps for compiling firmware using the Docker Kitematic container. I updated the git config from the docker shell with "git config --global core.autocrlf false" per the instructions. When I run the docker i get the output below. 

I see that is is still erroring with "/usr/bin/env: ‘python\r’: No such file or directory" but not sure what is causing it still. 

I just installed docker and kitematic on my windows 10 machine and pulled a fresh clone of the repository.

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /opt/build/armbin/arm-none-eabi
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /opt/build/armbin/arm-none-eabi
-- The ASM compiler identification is GNU
-- Found assembler: /opt/build/armbin/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /havoc/build
Scanning dependencies of target baseband_shared
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/crt0.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios-portapack/os/ports/GCC/ARMCMx/LPC43xx_M4/vectors.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore_v7m.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/common/ARMCMx/nvic.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsys.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdebug.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chlists.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chvt.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chschd.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chthreads.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdynamic.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chregistry.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsem.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmtx.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chcond.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chevents.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmsg.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmboxes.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chqueues.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmemcore.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chheap.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmempools.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/test.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testthd.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testsem.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testmtx.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testmsg.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testmbox.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testevt.c.obj
[  5%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testheap.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testpools.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testdyn.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testqueues.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/test/testbmk.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/hal.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/adc.c.obj
[  6%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/can.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/ext.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/gpt.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/i2c.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/icu.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/mac.c.obj
[  7%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/mmc_spi.c.obj
[  8%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/mmcsd.c.obj
[  8%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/pal.c.obj
[  8%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/pwm.c.obj
[  8%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/rtc.c.obj
[  8%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/sdc.c.obj
[  8%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/serial.c.obj
[  9%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/serial_usb.c.obj
[  9%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/spi.c.obj
[  9%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/tm.c.obj
[  9%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/hal/src/uart.c.obj
[... 59 lines cut ...]
[ 19%] Building CXX object firmware/baseband/CMakeFiles/baseband_shared.dir/tone_gen.cpp.obj
[ 19%] Built target baseband_shared
Scanning dependencies of target baseband_no_operation.elf
[ 20%] Building CXX object firmware/baseband/CMakeFiles/baseband_no_operation.elf.dir/proc_noop.cpp.obj
[ 20%] Linking CXX executable baseband_no_operation.elf
[ 20%] Built target baseband_no_operation.elf
Scanning dependencies of target libopencm3_hackrf_usb
[ 20%] Creating directories for 'libopencm3_hackrf_usb'
[ 20%] No download step for 'libopencm3_hackrf_usb'
[ 21%] No patch step for 'libopencm3_hackrf_usb'
[ 21%] No update step for 'libopencm3_hackrf_usb'
[ 21%] No configure step for 'libopencm3_hackrf_usb'
[ 21%] Performing build step for 'libopencm3_hackrf_usb'
  GENHDR  include/libopencm3/efm32/efm32g/irq.yaml
/usr/bin/env: ‘python\r’: No such file or directory
Makefile:58: recipe for target 'include/libopencm3/efm32/efm32g/irq.yaml.genhdr' failed
make[4]: *** [include/libopencm3/efm32/efm32g/irq.yaml.genhdr] Error 127
hackrf/firmware/hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/build.make:110: recipe for target 'hackrf/firmware/hackrf_usb/libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build' failed
make[3]: *** [hackrf/firmware/hackrf_usb/libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build] Error 2
CMakeFiles/Makefile2:397: recipe for target 'hackrf/firmware/hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/all' failed
make[2]: *** [hackrf/firmware/hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/all] Error 2
CMakeFiles/Makefile2:692: recipe for target 'firmware/CMakeFiles/firmware.dir/rule' failed
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
Makefile:313: recipe for target 'firmware' failed
make: *** [firmware] Error 2

stephen3303, 2021-08-11:
I have resolved the problem I was having. I did not realize that I cloned the repository prior to setting the ""git config --global core.autocrlf false' variable. If anyone finds themselves in this same circumstance just delete and re-clone the repository after setting the variable.

closed 2021-08-11 by stephen3303

eried, 2021-08-11:
Oh, yes, the line ending problem is quite annoying