portapack-mayhem/mayhem-firmware #405: Can't Load ADS-B Map After Build

The-Doctor-Of-11, 2021-08-13:
After building the world map bin using generate_world_map.py, the map will not load when opening an aircraft in the menu. Shows half black screen, red triangle, and half grey static.

**To Reproduce**
Steps to reproduce the behavior:
1. Load firmware
2. Build map using python3 and generate_world_map.py
3. Open ADS-B Reciever
4. Select aircraft
5. Select "See On Map"

I would expect this to show me a world or local map of where the aircraft is currently located / facing

A video and still image are attached in the Imgur link here: https://imgur.com/a/N7t1vdN 

Thank you!

eried, 2021-08-13:
Does it work ok with the provided map?

The-Doctor-Of-11, 2021-08-13:
> Does it work ok with the provided map?

Yes that fixed it, thank you!
Something is wrong with either the way I made the file or the python itself but for now it works, thank you!

closed 2021-08-13 by The-Doctor-Of-11