portapack-mayhem/mayhem-firmware #408: Problem upgrading the firmware

joelsmithlowe, 2021-08-30:
Upgrading from mayhem 1.2 to 1.4 bricked my Hack RF One Portapack. I cannot restore the original firmware just constant blank screen after following the upgrade guide and DFU mode is not allowing the device to be flashed spiflash cannot find the hack rf after doing the reset procedure. Not sure what I did wrong but 1.2 was easier to flash.

taclane, 2021-08-30:
If you have older versions of the hardware, give this a try.  
https://github.com/eried/portapack-mayhem/issues/346#issuecomment-890276806

I got caught in the same black screen loop with an early-model HackRF and Portapack H1. After re-flashing`hackrf_one_usb.dfu`in DFU mode, jslcom's 1.4 firmware worked fine after a hackrf_spiflash.

joelsmithlowe, 2021-08-31:
Thank you I am taking another look using my linux OS to compile as not having much luck in Windows

joelsmithlowe, 2021-09-02:
I ended up getting 1.3.1 flashed onto the device and is working again. Will attempt to flash to 1.4 again.

I have added the below notes in case it is of use to anyone.

1. (Device Shows as below initially)
Bus 001 Device 005: ID 1fc9:000c NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)

2. dfu-util --device 1fc9:000c --alt 0 --download hackrf_usb.dfu

3. hackrf_spiflash -w hackrf_one_usb.bin
- If error make sure cd to dir where .bin file is located.

4. (Device Shows as below identifier)
Bus 001 Device 009: ID 1d50:6089 OpenMoko, Inc. Great Scott Gadgets HackRF One SDR

5. cd to hackrf folder and run hackrf_cpldjtag -x firmware/cpld/sgpio_if/default.xsvf

6.  Remove/Reconnect USB, run hackrf_info to confirm device is detected and run hackrf_spiflash -w portapack-h1_h2-mayhem.bin
- Same as above if error make sure cd to dir where .bin file is located.

7. Once Writing 873764 bytes at 0x000000 displays wait about 30 seconds to be sure and remove/reconnect USB. Portapack should boot as expected.

closed 2021-09-02 by eried