portapack-mayhem/mayhem-firmware #410: Auto frequency control for WFM in 5mhz Bandwight

Egho9, 2021-08-31:
will be very great if can add function which can auto frequency control for WFM in 5mhz bandwight

gullradriel, 2023-05-20:
What is that thing?
@Brumi-2021 can you enlighten me ?

Brumi-2021, 2023-05-20:
Hello @Egho9 and @gullradriel , i know that some SDR programs has similar feature, but I guess that in less bandwith.  

In commercial WFM broadcasting band , it could be stations every 200khz (sometimes even closer)  , then inside 5 Mhz it could be many stations,  and the function can get crazy . 

In satellite FM  reception, it could be usefull to compensate the +/- doppler effect , but based on my  experience , that use to be +-5khz tracking aprox  ...  

Anyway it would be nice to understand, in  which program did you see  that feature and what could be the practical  use case ? 

Cheers