portapack-mayhem/mayhem-firmware #411: Three flashing LEDs after reset.

Noname12345GER, 2021-08-31:
**Describe the bug**
During normal Operation, the Portapack froze. I pressed the Reset Button on the HackRF. After Reset: 

1. The screen stays black.
2. HackRF LEDs (The 3 rightmost, USB, RX, TX) are blinking together at the same time.

**To Reproduce**
Im sorry, i do not know.

**Expected behavior**
Well, i expected the Portapack to turn on normally.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* The Latest Stable Release

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

I thought the whole unit was fried because Windows did not recognize it anymore. I did a succsessful DFU recovery, but it just reverts to 3 leds flashing.

If i remove the portapack from the HackRF, The HackRF gets detected by Windows again and seems to operate normally.
I have looked thru the HackRF and the Portapack Wiki for the 3 LED symtom, but no luck.

eried, 2021-09-02:
Have you tried the previous version? 1.3.1?

Noname12345GER, 2021-09-02:
I will try.

Noname12345GER, 2021-09-03:
No, does not work.

jeniceek, 2021-09-29:
It helps to connect and disconnect USB to make it work.

euquiq, 2021-10-03:
Maybe it is a hardware problem ? (regrettable, portapack / hackRF is broken) ? 

Probably a long shot: I would try opening the case, exposing the hackRF and Portapack boards. Then I would separate them, visually inspecting each one and even cleaning if necessary. Maybe something metallic ended inside your portapack and it is creating a short  somewhere ?

linuxgorl, 2021-10-04:
It does the same, after unplugging my H2 when its internal battery still holds a charge. Unplugging and replugging the battery makes the portapack work again. It dosen't happen when I unplug it when it is in HackRF mode, so maybe there is just missing code in the USB handler

ArjanOnwezen, 2021-11-18:
Looks like a device specific issue.

closed 2021-11-18 by ArjanOnwezen

deepuj, 2022-05-01:
Any luck with this? I started seeing this today after updating to Mayhem 1.5.1 (H2+).

Was receiving fine for a few minutes, and the screen went off. I pressed the reset button and Tx/Rx/USB started blinking. Disconnecting the battery helped it reset and work properly. But it seems each time I reset this happens where i have to disconnect and reattach the battery.

GullCode, 2022-05-01:
When it happens just turning it off and on does the trick on mine, without
taking out the battery.

jamjahal, 2022-05-03:
I am seeing the same behavior.  Unplugging / replugging has not helped.  

Running `hackrf_info` returns 
"""
hackrf_info version: unknown
libhackrf version: unknown (0.6)
No HackRF boards found.
"""

Have tried switching out the usb cable, plugging into different USB ports, but no luck.

eried, 2022-05-04:
@jamjahal your problem might be related with https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting check the first video at the end of that page

deepuj, 2022-05-04:
Some wired sequence of pressing the buttons helps me. Any reset by pressing the reset button potentially puts the device in the blank screen and 3 LEDs flashing mode. 

I tried going to and.back out of DFU mode a few times and it helps come out of the flashing LEDs. Still no screen (but buttons respond). Some times pressing the left button while resetting helps, some times some random sequence of keys.

In the screen blank mode, I tried looking at the lcd with a bright light to make sure it is not the back light being off. But seems there is nothing on the LCD. May be the LCD driver is not initializing at all due to missed events?