portapack-mayhem/mayhem-firmware #414: RDX not sending proper text

tikky, 2021-10-03:
Whatever text I am trying to send via RDX/FM always "test123" is sent, not my own text.
Looks like a bug and hardcoded text in firmware.

euquiq, 2021-10-03:
I search this repository for "test123" and there is no mention for that string, other than this text in your issue. Are you sure that you are spelling correctly (lower case, test123) ?

What firmware exactly are you using ?

On the other hand, I can find a reference for the string "TEST1234" on apps/ui_rds.hpp  ...  would it be on TX -> RDS app, when you set the PSN ?

tikky, 2021-10-04:
Yes, sorry, it is always "TEST1234" even if I set different text and click "transmist radio text".
Firmware is 1.4.0

eried, 2021-10-04:
Hey, thanks for the report :) 

@euquiq I find sweet that the code is finally searchable here! 😂 https://github.com/search?q=user%3Aeried+test1234&type=code

tikky, 2021-10-04:
Glad to know that the problem is found :-)
When we can expect new version with working RDX text?

euquiq, 2021-10-04:
Problem is: The "TEST1234" string is for the PSN name ... not the "text" sent (on the text tab) ... having said that, you made me search all the house in order to find an RDS (not RDX) radio ...

It works fine.

You really need to experiment a bit more. As said above, you are confusing the PSN with the "radio text".

euquiq, 2021-10-04:
> Hey, thanks for the report :)
> 
> @euquiq I find sweet that the code is finally searchable here! 😂 https://github.com/search?q=user%3Aeried+test1234&type=code

Yes! but it does not search partial texts (so you erase that "4" and it cannot find the test123).

ArjanOnwezen, 2021-11-18:
@tikky psn can be changed on the first tab, using the SET button. TEST1234 is just the default value.

closed 2021-11-18 by ArjanOnwezen