portapack-mayhem/mayhem-firmware #420: Problems compiling 1.4.1 on MacOS using multiple versions of gcc-arm-none-eabi

sck-nogas, 2021-11-02:
**Describe the bug**
Getting errors when compiling 1.4.1 on same system/compiler that worked with 1.4.0...

**To Reproduce**
Steps to reproduce the behavior:
1. git pull
2. mkdir build
3. cd build
4. cmake ..
5. make firmware

**Expected behavior**
It compiles? :) 

**Additional write with output of three different compiles using three different version of the compiler**
On my iMac using gcc-arm-none-eabi-9-2020-q2-update and cmake version 3.21.4 on MacOS 11.6, which previously compiled 1.4.0 correctly I p[ulled the updates tot eh repo and did a rebuild and got this error...

iBiko:portapack-mayhem sck$ git pull && rm -rf build && git submodule update --init --recursive && export PATH=~/bin/gcc-arm-none-eabi-9-2020-q2-update/bin:${PATH} && mkdir build && cd build && cmake .. && make firmware && hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin
Already up to date.
CMake Deprecation Warning at CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/sck/src/portapack-mayhem/build
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/crt0.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios-portapack/os/ports/GCC/ARMCMx/LPC43xx_M4/vectors.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore_v7m.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/common/ARMCMx/nvic.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsys.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdebug.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chlists.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chvt.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chschd.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chthreads.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdynamic.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chregistry.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsem.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmtx.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chcond.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chevents.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmsg.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmboxes.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chqueues.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmemcore.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chheap.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmempools.c.obj
[... 394 lines cut ...]
/Users/sck/src/portapack-mayhem/firmware/baseband/dsp_goertzel.cpp:44:6: warning: array subscript 2 is above array bounds of 'int16_t [2]' {aka 'short int [2]'} [-Warray-bounds]
   44 |   s[2] = s[1];
      |   ~~~^
In file included from /Users/sck/src/portapack-mayhem/firmware/baseband/dsp_goertzel.cpp:23:
/Users/sck/src/portapack-mayhem/firmware/baseband/dsp_goertzel.hpp:38:10: note: while referencing 'dsp::GoertzelDetector::s'
   38 |  int16_t s[2] { 0 };
      |          ^
[ 15%] Building CXX object firmware/baseband/CMakeFiles/baseband_shared.dir/matched_filter.cpp.obj
[ 15%] Building CXX object firmware/baseband/CMakeFiles/baseband_shared.dir/spectrum_collector.cpp.obj
/Users/sck/src/portapack-mayhem/firmware/baseband/spectrum_collector.cpp: In instantiation of 'typename T::value_type spectrum_window_hamming_3(const T&, size_t) [with T = std::array<std::complex<float>, 256>; typename T::value_type = std::complex<float>; size_t = unsigned int]':
/Users/sck/src/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:145:79:   required from here
/Users/sck/src/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:116:28: error: non-constant condition for static assertion
  116 |  static_assert(power_of_two(s.size()), "Array size must be power of 2");
      |                ~~~~~~~~~~~~^~~~~~~~~~
/Users/sck/src/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:116:28: error: 's' is not a constant expression
/Users/sck/src/portapack-mayhem/firmware/baseband/spectrum_collector.cpp:117:19: error: 's' is not a constant expression
  117 |  constexpr size_t mask = s.size() - 1;
      |                   ^~~~
make[3]: *** [firmware/baseband/CMakeFiles/baseband_shared.dir/spectrum_collector.cpp.obj] Error 1
make[2]: *** [firmware/baseband/CMakeFiles/baseband_shared.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2
iBiko:build sck$ 

Same compile error as previously. Any suggestions on where to start?

sck-nogas, 2021-11-02:
Forgot to include....

```
iBiko:portapack-mayhem sck$ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	firmware/tools/airlines.db
	sdcard/ADSB/airlines.txt

no changes added to commit (use "git add" and/or "git commit -a")
iBiko:portapack-mayhem sck$ 
```
And
```
iBiko:portapack-mayhem sck$ cmake --version
cmake version 3.21.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
iBiko:portapack-mayhem sck$ 
```

Brumi-2021, 2021-11-02:
Hello are you compiling with gcc version 10 ? 
The current version 1.4.1 still shows the same   compiling errors than Mayhem 1.4.0  with gcc version 10 . You should wait till next version release Mayhem 1.4.2 (coming soon) or apply manually my PR , https://github.com/eried/portapack-mayhem/pull/419

sck-nogas, 2021-11-02:
I tried to compile with version 9.3.1 , version 10.2.1, and version 10.3.1.

Brumi-2021, 2021-11-02:
Hi , and  when you tried to compile , 9.3.1 , you also followed all the listed steps , (https://github.com/eried/portapack-mayhem/wiki/Compile-firmware), including that one : 

5. Replace the python version in libopencm3 to use python3 (as user)

sed -i 's/env python/env python3/g' /opt/portapack-mayhem/hackrf/firmware/libopencm3/scripts/irq2nvic_h

sck-nogas, 2021-11-02:
Don't think that will change anything, since I'm on MacOS and not Debian, and that...
```
iBiko:~ sck$ python -V
Python 3.7.12
iBiko:~ sck$ python3 -V
Python 3.7.12
iBiko:~ sck$ 
```
But, I'm game to try...
```
Biko:portapack-mayhem sck$ grep "env python" hackrf/firmware/libopencm3/scripts/irq2nvic_h 
#!/usr/bin/env python
iBiko:portapack-mayhem sck$ gsed -i "s/env python/env python3/g" hackrf/firmware/libopencm3/scripts/irq2nvic_h 
iBiko:portapack-mayhem sck$ grep "env python" hackrf/firmware/libopencm3/scripts/irq2nvic_h 
#!/usr/bin/env python3
iBiko:portapack-mayhem sck$ 
```
Different failure this time...
```
iBiko:portapack-mayhem sck$ git pull && rm -rf build && git submodule update --init --recursive && export PATH=~/bin/gcc-arm-none-eabi-9-2020-q2-update/bin:${PATH} && mkdir build && cd build && cmake .. && make firmware && hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin
Already up to date.
CMake Deprecation Warning at CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: /Users/sck/bin/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/sck/src/portapack-mayhem/build
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/crt0.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios-portapack/os/ports/GCC/ARMCMx/LPC43xx_M4/vectors.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/GCC/ARMCMx/chcore_v7m.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/ports/common/ARMCMx/nvic.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsys.c.obj
[  1%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdebug.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chlists.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chvt.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chschd.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chthreads.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chdynamic.c.obj
[  2%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chregistry.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chsem.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmtx.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chcond.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chevents.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmsg.c.obj
[  3%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmboxes.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chqueues.c.obj
[  4%] Building C object firmware/baseband/CMakeFiles/baseband_shared.dir/__/chibios/os/kernel/src/chmemcore.c.obj
[... 184 lines cut ...]
  GENHDR  include/libopencm3/sam/3n/irq.yaml
  GENHDR  include/libopencm3/lpc17xx/irq.yaml
  GENHDR  include/libopencm3/lpc13xx/irq.yaml
  GENHDR  include/libopencm3/lpc43xx/m0/irq.yaml
  GENHDR  include/libopencm3/lpc43xx/m4/irq.yaml
  GENHDR  include/libopencm3/stm32/f4/irq.yaml
  GENHDR  include/libopencm3/stm32/f3/irq.yaml
  GENHDR  include/libopencm3/stm32/f2/irq.yaml
  GENHDR  include/libopencm3/stm32/f0/irq.yaml
  GENHDR  include/libopencm3/stm32/l1/irq.yaml
  GENHDR  include/libopencm3/stm32/f1/irq.yaml
  GENHDR  include/libopencm3/lm3s/irq.yaml
  GENHDR  include/libopencm3/efm32/efm32gg/irq.yaml
  GENHDR  include/libopencm3/efm32/efm32g/irq.yaml
[... cut ...]
make[5]: *** No rule to make target `/Users/sck/bin/gcc-arm-none-eabi-9-2019-q4-major/lib/gcc/arm-none-eabi/9.2.1/include/stdint.h', needed by `flash.o'.  Stop.
make[4]: *** [lib/stm32/f0] Error 2
make[3]: *** [hackrf/firmware/hackrf_usb/libopencm3_hackrf_usb-prefix/src/libopencm3_hackrf_usb-stamp/libopencm3_hackrf_usb-build] Error 2
make[2]: *** [hackrf/firmware/hackrf_usb/CMakeFiles/libopencm3_hackrf_usb.dir/all] Error 2
make[1]: *** [firmware/CMakeFiles/firmware.dir/rule] Error 2
make: *** [firmware] Error 2
iBiko:build sck$ 
```

Brumi-2021, 2021-11-04:
Hi,  now, as you said ,  different compiler errors, getting closer !    
I suggest  you,   to take care one by one of the errors,. ex,  that one  ,  *** No rule to make target `,  

In my case, I am also not using  Debian (same as you) , but eventhough , I am using Gentoo distro and it  is also  linux   ,  therefore, I followed all the steps that I need..  basically  from the (3) onwards,  checking  carefully  one  by one ,if they had sense or not, and  if the  directory path folders were exactly the same or not ...  And eventually I am compiling it with  gcc-10.2.0 very fast , withouth any problems. (just  adding  and merging  my commented  PR)

Then same advise , pls  follow  previous Mayhem  compile  wiki guide , and I hope and believe that  you should  be able to compile Mayhem  with gcc 9 without any error . (even some  warnings, that you can ignore ) .
 
If not , we should better  ask additional  help  , from other more advanced linux Mac user   help .

sck-nogas, 2021-11-05:
Okay, this was strange (_**but it's working now!**_)...

On my Mac, the 'git status' showed that I was in sync and had no "dangling bits". **BUT**, there was some dangling cmake history cruft somewhere since I was compiling with "_gcc-arm-none-eabi-9-**2020-q2-update**_" but the error was it was trying to include things from "_gcc-arm-none-eabi-9-**2019-q4-major**_"...

So, this was the working sequence to compile on MacOS using gcc-arm-none-eabi-9-2020-q2-update for me... 

1. `cd src`   # or wherever you keep the sourcecode for this...
2. `rm -rf portapack-mayhem`    # Blow away the Old directory 
3. `git clone --recurse-submodules https://github.com/eried/portapack-mayhem.git ` # Bring down the new stuff
4. `cd portapack-mayhem` # switch into the directory
5. `git checkout master`   # To switch from next to Master for version number update (otherwise it will still say version 1.4.0 on your portapack)
6. `export PATH=~/bin/gcc-arm-none-eabi-9-2020-q2-update/bin:${PATH}` # Add the compiler to my path
7. `cd firmware` 
8. `rm -rf build && mkdir build && cd build`   # Make sure build dir is clean
9. `cmake -B./ -S../../ ` # do the cmake 
10. `make firmware` # make
11. `hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin` # copy the firmware over

Still can't compile with gcc-arm-none-eabi-10.3-2021.10 though. But, did not expect it to, based on Brumi's comments/PR.

closed 2021-11-05 by sck-nogas