portapack-mayhem/mayhem-firmware #427: After updating firmware the screen touch function is ok but displays nothing

Jazzzrory, 2021-11-08:
Hello, 

After I updated the firmware (ver 1.4.1), the switch was normal, the program ran normally, the **screen** touch function was ok but **displayed nothing**. 

（Repressing the Reset key is still the same. No program has been modified.）

I have never met this before, if you know the solution, please let me know as soon as possible. 

Thank you very much！

eried, 2021-11-08:
Was it ok with the previous 1.4.0?

Jazzzrory, 2021-11-09:
Yes，it was ok with 1.4.0.

ArjanOnwezen, 2021-11-09:
What does exactly happen. Does it show the splash screen and menu.

eried, 2021-11-09:
> Yes，it was ok with 1.4.0.

This is very strange. Can you record a video with 1.4.0 and then 1.4.1 showing the issue on the later?

ArjanOnwezen, 2021-11-24:
@Jazzzrory  This looks like a device specific issue. Without extra info impossible to determine the cause. Please try 1.4.2

closed 2021-11-24 by ArjanOnwezen