portapack-mayhem/mayhem-firmware #431: World Map Generation is 8gb, fat32 only does 4

greggawatt, 2021-11-16:
**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Tap on '....'
3. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release
* Previous one (if any) that did not presented the issue
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

ArjanOnwezen, 2021-11-16:
@greggawatt  please fill in the placeholder

eried, 2021-11-17:
For the current implementation your generated map should be smaller than 4 GB, so you should reduce the input file by half

greggawatt, 2021-11-17:
I'm using the input file from this repo, is there another jpg I can use for reference?

saormart, 2021-11-17:
I had the same issue using the image file from this repo!

eried, 2021-11-18:
Oh, you both are right, then the provided file is not the right one. I will fix it soon, I have no idea why I placed a bigger map instead of the one I used.

May I ask why you want to create your own file instead of the ready to use from the package?

saormart, 2021-11-22:
> Oh, you both are right, then the provided file is not the right one. I will fix it soon, I have no idea why I placed a bigger map instead of the one I used.
> 
> May I ask why you want to create your own file instead of the ready to use from the package?

I'm using now from the package, but I was just learning and testing how to do it on my own using the repo image, when I figure out that it was creating a large file that didn't fit on my SD-Card... I though that I made some mistake until @greggawatt open this ticket and I decided to comment that I had the same issue!

ArjanOnwezen, 2022-04-08:
@eried do you have the file somewhere still available. Else it might be a good idea to remove the file to avoid confusion..

eried, 2022-04-08:
Ok I need to check this after holidays 🧐

ArjanOnwezen, 2022-04-08:
Great.. Happy Holidays!

eried, 2022-05-02:
Hey @greggawatt and @saormart I was checking this issue, sorry for the delay, but now I am a bit confused on what is happening,  because the generation should use 2 bytes per pixel as shown here: https://github.com/eried/portapack-mayhem/blob/18409e6747ccd84bb8e7b0828cf5b1407bd3c28d/firmware/tools/world_map.py#L39 so that would mean that the jpg needs to be 32768 to be the same size as the file included in the release package (2 GB).

I uploaded the source files I used https://mega.nz/folder/StUQ0bZL#1hsIbP34TQku07OYa5YlTQ (the giant photoshop file used to combine all the layers from the tiles I downloaded) but I am still thinking why you both are getting a 8 GB one...

MattLodge, 2022-05-03:
So I was having a similar issue, but I think I've resolved it. I'm using python 3.7.3.
If I used the most recent script & JPG in the master branch, I got the 8gb file. I modified the script to remove strings and use bytes directly, and have got a 2gb binary out. I haven't done any tested this on the portapack yet though
```
for y in range (0, im.size[1]):
	line = b''
	for x in range (0, im.size[0]):
		# RRRRRGGGGGGBBBBB
		pixel_lcd = (pix[x, y][0] >> 3) << 11
		pixel_lcd |= (pix[x, y][1] >> 2) << 5
		pixel_lcd |= (pix[x, y][2] >> 3)
		#         RRRGGGBB to
		# RRR00GGG000BB000
		# pixel_lcd = (pix[x, y][0] >> 5) << 5
		# pixel_lcd |= (pix[x, y][1] >> 5) << 2
		# pixel_lcd |= (pix[x, y][2] >> 6)
		line += struct.pack('H', pixel_lcd)
	outfile.write(line)
	print(str(y) + '/' + str(im.size[1]) + '\r', end="")
```

eried, 2022-05-03:
Nice, please report your results

MattLodge, 2022-05-07:
It appears to work for me. I'll prep a pull request this weekend

jLynx, 2022-05-08:
Merged in

closed 2022-05-08 by jLynx