portapack-mayhem/mayhem-firmware #435: Next is behind master etc...

loewal, 2021-11-23:
Hello Erwin,

I have 3 questions about things I don't quite understand:
1) a few hours ago there was a commit of adsb into next. I would like to try that, but then I saw that next is more then 200 commits behind master....
2) did you stop compiling, because the latest nightly build is quite old?
3) when I do a git clone..., I see that the url is always the same, next or master, no difference.

I would like the v1.4.1 with the adsb-enhancement.
How to obtain?

Thanx!!

eried, 2021-11-23:
hi,

> 1. a few hours ago there was a commit of adsb into next. I would like to try that, but then I saw that next is more then 200 commits behind master....

200 behind sharebrained's master, not our master (which is normal)

> 2\. did you stop compiling, because the latest nightly build is quite old?

the service I was using migrated the project and somehow deleted the script that compiled the nightly. So I havent made a new one yet

> 3\. when I do a git clone..., I see that the url is always the same, next or master, no difference.

this is handled by git, with `git checkout <existing_branch>`

loewal, 2021-11-23:
Thanks Erwin,
I compiled next: got v.14.0
I dowload the zip-file and populated the hakrf folder etc. and compiled: still v1.4.0
And yes, all done within the next-branch.
Where do I go ...wrong?

loewal, 2021-11-23:
Shouldn't that be v1.4.1??

eried, 2021-11-23:
Nothing, it is a sequence error from my side (I was merging next into master before updating the version number, so next always stayed in the older version). If you built from next then you have the latest possible.

ArjanOnwezen, 2021-11-24:
Solved with new release today

closed 2021-11-24 by ArjanOnwezen