portapack-mayhem/mayhem-firmware #440: Problem upgrading the firmware

aesaganda, 2021-11-29:
I have a HackRF One replica with Portapack H2 extension. The device shipped with Mayhem v.1.4.0 firmware and worked as expected. But when I flashed v.1.4.2 firmware with flash_portapack_mayhem.bat method, screen became unresponsive even after multiple reboots. So I have proceed to go with dfu_hackrf_one.bat method. I have followed steps provided in the tutorial. 
**After I put the device in DFU mode 
an error message like this showed up.**

[image]

**After multiple trials I give up hope on Windows and tried same steps with my Kali installed Raspberry Pi.**

[image]

**Same error message showed up again.** 

In Windows 11 device manager I cannot see any HackRF One or LPC device. 
[image]

**What should I do in order to unbrick my device? Thanks in advance.**

TheDragonkeeper, 2021-11-30:
Can you post the output of "dmesg" when you plug the device in with hackrf mode?

i seem to have the same issue on my real hackrf one but on fw 1.4.1    but suspect a different issue my end.

aesaganda, 2021-11-30:
When I first tried flashing the firmware, the portapack daughterboard was attached to HackRF. I tried dmesg and lsusb device couldn't recognized by PC.

[image]
 
**After a couple of unsuccessful attempts I decided to try detach portapack, thus leaving HackRF board bare. With this trick I have managed to get DFU mode and get device listed on lsusb and dmesg.**

[image]

[image]

It also got listed in Windows Device Manager as LPC device too.

After I managed to bring DFU mode on I proceeded to follow the guide on README and flash dfu_hackrf_one.bat. This step makes HackRF firmware run on board's RAM and after this step PC recognizes it no longer as LPC device but HackRF One device in Windows Device Manager. After this step I have flashed flash_portapack_mayhem.bat in order to make it persistent. Command line output seems successful, but after rebooting device screen goes blank (it doesn't even have backlight) and unresponsive.
[image]
[image]

**I have also tried flashing v1.4.0 (original fw seller provided) but got the same results.**

TheDragonkeeper, 2021-11-30:
Thanks. unfortunatly mines not the same issue.  mine must be due to physical modifications, which seemed to give me one good use to flash 1.4.1  and hasnt worked since on pc, i can only use mine via portapack now.

TheDragonkeeper, 2021-11-30:
thought id post way i fix my blank screen issue tho just incase it helps:
hit the reset button
double click the dial (the green lights should go off)
hold the dial down and then hold the up button 
release the up button then release the dial
screen comes on

aesaganda, 2021-11-30:
Tried the steps but I had no luck. Thanks anyway.

gadjodilo83, 2021-11-30:
try using a different USB cable or flashing without the portapack...

aesaganda, 2021-11-30:
Changing USB cable didn't help. I flashed without the portapack, because it seems its the only way to get the board listed in the device manager; since the touchscreen is unresponsive and couldn't get into hackrf mode. But as I stated above that didn't work as well.

lyusupov, 2021-12-01:
PP H2+ (with HackRF one and Li-Po battery connected) f/w update instructions for Raspberry Pi:

1) disconnect H2 USB cable from RPi
2) power off H2 by double click on encoder button
3) press and keep holding HackRF One DFU button
4) connect H2 to Raspberry PI

pi@raspberrypi: $ lsusb | grep NXP
Bus 001 Device 070: ID 1fc9:000c NXP Semiconductors 

5) upload HackRF One DFU (RAM) firmware at first

pi@raspberrypi: $ dfu-util -D hackrf_one_usb.dfu
< ... skipped ...>
pi@raspberrypi:/run/tmp/PP $ lsusb | grep Open
Bus 001 Device 071: ID 1d50:6089 OpenMoko, Inc. 

6) upload Mayhem firmware next:

pi@raspberrypi: $ hackrf_spiflash -w portapack-h1_h2-mayhem.bin
File size 919828 bytes.
Checking target device compatibility
Erasing SPI flash.
Writing 919828 bytes at 0x000000.

7) disconnect and re-connect USB cable to reset the PP

aesaganda, 2021-12-01:
[image]

Even after steps mentioned above, when I click reset button or reboot the device, firmware cannot be read from SPI flash. The LEDs on top of the board get dull. The device can only be used in dfu mode (firmware in ram) but still no luck reviving portapack with Mayhem FW. I started to think that board's flash memory is corrupted, therefore I cannot flash via SPI, even it shows successful output at CLI.

lyusupov, 2021-12-01:
> I started to think that board's flash memory is corrupted,

Use _hackrf_spiflash_ to read your SPI flash IC content,

```
pi@raspberrypi:~ $ hackrf_spiflash 
Specify either read, write, or reset option.
Usage:
	-h, --help: this help
	-a, --address <n>: starting address (default: 0)
	-l, --length <n>: number of bytes to read (default: 1048576)
	-r, --read <filename>: Read data into file.
	-w, --write <filename>: Write data from file.
	-i, --no-check: Skip check for firmware compatibility with target device.
	-d, --device <serialnumber>: Serial number of device, if multiple devices
	-s, --status: Read SPI flash status registers before other operations.
	-c, --clear: Clear SPI flash status registers before other operations.
	-R, --reset: Reset HackRF after other operations.
	-v, --verbose: Verbose output.
```

 then make a binary comparison with the file you want to flash in.

lyusupov, 2021-12-01:
https://github.com/greatscottgadgets/hackrf/search?q=hackrf_spiflash&type=issues

aesaganda, 2021-12-01:
The issue is same as mine so I decided to follow the steps.
But when I run the commands, the board doesn't responded at all so I interrupted it with keyboard.

> hackrf_spiflash -s -v output:

[image]

> hackrf_spiflash -c output:

[image]

So I am still unable to erase the data on flash chip.

aesaganda, 2021-12-01:
Ok. I finally solved the issue.
After fiddling up with Hackrf host tools I found apt repository's are not the latest version. So I removed them and
[https://github.com/greatscottgadgets/hackrf/tree/master/host](url) from this link I build host tools from scratch. After these steps I have managed flashing the latest firmware successfully. Thanks for everyone helping out. I couldn't have revived my device without you.

closed 2021-12-01 by aesaganda