portapack-mayhem/mayhem-firmware #442: Frequency offset option for upconverter?

chrismetcalf, 2021-12-02:
I'm loving my HackRF + PortaPack, but as a ham radio operator I'd love to be able to use features like Looking Glass to explore the HF bands while mobile with an upconverter like the Ham It Up. 

However, I can't find an option to set an offset to allow Mayhem to know it's hooked up to an upconverter and apply a configurable frequency offset.

**Describe the solution you'd like**

- An option in the Settings menu, probably under "Radio" to configure an upconverter offset
- The ability to toggle that offset on and off, either from that same menu or even better from the top menubar

**Describe alternatives you've considered**

I guess I could just apply the offset mentally, but that's no fun.

Dirk275, 2021-12-02:
Agreed.
And a frequency calibration option would be fine too.

Brumi-2021, 2021-12-02:
Hi , I am also ham radio operator , and sometimes I already enjoyed a lot , with Hackrf + 200Mhz (LF/MF/HF) up converter  and Youloop antenna ,very nice portable solution. But in my case the offset is easy. But good suggested feature .👍

gullradriel, 2023-05-10:
Converter feature was added.
Check the wik for informations on the app.
Available on last stable and nightly.

closed 2023-05-10 by gullradriel