portapack-mayhem/mayhem-firmware #443: Center button and dial issues

Reaper-Oni, 2021-12-03:
**Describe the bug**
During use of the portapack the centre button functionality sometimes doesn’t work. With capturing and trying to oom in on the signal the button doesn’t work and zooming in isn’t possible. 

Also when trying to edit time the center button only functions as and enter and changing time isn’t possible 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to capture
2. Tap capture and start and try to change 500k
3. Changing the 500k isn’t possible

4. Go to setting
5. Tap time and date settings and try to change time and date
6. Changing the time and date isn’t possible

**Expected behavior**
Zooming on signal to inspect activity. 
changing time and date settings

**Affected versions**
Please write any difference related with the Expected behavior, on the following versions:
* Latest Stable Release
* Previous one (if any) that did not presented the issue
* Old versions available here: https://github.com/eried/Research/tree/master/HackRF/PortaPack/old_legacy_firmware

**Additional**
If the bug is difficult to explain, additionally to the text please include images and videos.

ArjanOnwezen, 2021-12-03:
@Reaper-Oni  I can't Reproduce the issue. I can change date & time and also change the 500k in capture app, using 1.4.2

Reaper-Oni, 2021-12-03:
@ArjanOnwezen then please tell me how to change the settings.

ArjanOnwezen, 2021-12-03:
Just select the date and time (move there using buttons) then use the dial to select other values. For time move to top left first. Then move down.

Reaper-Oni, 2021-12-03:
@ArjanOnwezen reloaded the software and all seems to be working now. Thanks for the pointers.

closed 2021-12-03 by Reaper-Oni