portapack-mayhem/mayhem-firmware #444: Adjust Rotary Wheel Sensitivity?

bluegizmo83, 2021-12-04:
**Describe the bug**
My new Portapack H2 seems to have a slight issue with the rotary wheel. It moves twice for every single indent of the wheel. Makes it a pain to use. I'm quite sure this is an issue with the hardware, not the firmware, but is there any way I could change something in the firmware and recompile to make it less sensitive or ignore every other input when turning the wheel?

**Expected behavior**
Turning the rotary wheel should only send one input per indentation of the encoder. Currently sends two.

**Affected versions**
1.3.0 and 1.4.2 at least but like I said this seems to be hardware related since no one else seems to have this issue.

**Additional**

eried, 2021-12-06:
This can be added, but could you try first to clean the rotary wheel, i.e. with contact cleaner. I do not think this problem is very common like to change something globally.

bluegizmo83, 2021-12-07:
> This can be added, but could you try first to clean the rotary wheel, i.e. with contact cleaner. I do not think this problem is very common like to change something globally.

Ok I just tried cleaning it with electronics contact cleaner, and 90% isopropyl alcohol, neither of which helped. It's definitely a hardware issue though, and it's worse than I thought, as not only does it send two inputs for ever indentation of the encoder, but it's also sending opposite inputs occasionally as well! If I'm turning the encoder (fast or slow) to the right, it sends two right inputs but occasionally sends several left inputs as well and ends up jumping the opposite direction. It also does it when turning in the other direction as well (but sends the opposite direction occasionally too).

I'd say don't worry about this for now. It's not worth modifying firmware for one device having this issue. I think I'm just gonna try to find a replacement/similar encoder online and try to replace it myself.

saormart, 2021-12-07:
I have the same problem, and the problem is with the Rotary Wheel itself!
I contacted the seller of mine, and they're sending me a replacement due the batch get build, their vendor send them the wrong Rotary Wheel ....

bluegizmo83, 2021-12-07:
> I have the same problem, and the problem is with the Rotary Wheel itself! I contacted the seller of mine, and they're sending me a replacement due the batch get build, their vendor send them the wrong Rotary Wheel ....

Ah ok! Guess I'm not the only one experiencing this then. Sadly I'm outside the return window on mine so I'll have to look into replacing the encoder myself...

Did yours do both issues, double inputs and opposite inputs, or just double inputs?

bluegizmo83, 2021-12-07:
Just in case anyone is not sure what I'm talking about when I try to describe this problem, here is a short video I made showing what's happening:

https://youtu.be/zyu2_BEpYmw

eried, 2021-12-07:
> Video description explains what I'm doing in the video.

I see, that is clearly NOT how it normally works :) I know the software can help ease this issue (I have a 3d printer with a faulty encoder and I just use 2 "ticks" instead of 1 in the marlin firmware config file). So if you feel adventurous, you could try to modify the rotary encoder code of the firmware and post the fix here. If is a widespread issue, it would be worthy to add permanently, but as is for now I guess it would be useful for you two to have a workaround.

ArjanOnwezen, 2021-12-07:
Never seen such a "responsive" wheel before. 
Just try to get a refund or a replacement unit.

ArjanOnwezen, 2021-12-07:
@bluegizmo83  Ticket will be closed, since this seems a hardware issue fir a limited set of device.
@saormart  will get a replacement and hopefully you will too, after contacting seller.

closed 2021-12-07 by ArjanOnwezen

saormart, 2021-12-07:
> @bluegizmo83 Ticket will be closed, since this seems a hardware issue fir a limited set of device. @saormart will get a replacement and hopefully you will too, after contacting seller.

I think you should leave this open, and some how add the possible fix with an option to select if you have this Rotary Wheel type for those who may have this issue in the future, due I notice that most of "not official"  portapack may have this issue....
I don't have much time right now due a private project I'm working on, but whenever I have time I may able to help fix this and add as option to select the Rotary Wheel type or something like that!

ArjanOnwezen, 2021-12-07:
@saormart I'm not so sure. I mean creating an option for only a handful of devices. Best it to ask for a refund, so these faulty rotary wheels will go out of production.

bluegizmo83, 2021-12-08:
So I'm trying to mess with the firmware myself to see if I can fix this, but I'm not that familiar with C++ or how encoders work... I've narrowed my source code search down to these files:

encoder.cpp
irq_controls.cpp
event_m0.cpp

My first real question is, is there a way to test the firmware changes without having to compile and flash it to the device each time? Is there a way to load it temporarily into memory or something? Along with that, is there a way to debug it while its running on the device and plugged into my pc via USB? I would be nice if I could put in break points and see the actual output bit stream from the encoder as I'm turning it...

lu1aax, 2022-04-10:
I've just reported this issue and I don't think this is due to a faulty encoder but a different encoder being installed in the device. I could try to replace the encoder but I'd have to know what's the correct encoder to install.

lu1aax, 2022-04-10:
> 

It looks like we are three now!!! :) I'd like to contribute in fixing this... perhaps it's not that hard to add support to another model of encoder in the firmware, it's just about reversing directions and counting rotary clicks.

ArjanOnwezen, 2022-04-10:
It's not about hard or not to create an option in yhe software. It's also about usability. Too many options clutter the interface. Furthermore it makes developers life unnecessary harder. It's better to ask for a refund for incompatible devices, so they will no longer be sold.