portapack-mayhem/mayhem-firmware #445: Problem upgrading the firmware

Stein3247, 2021-12-04:
i have an issue after i have updated the portapack h2, it doesn't boot up, doesn't seem like the firmware that i have installed runs on my portapack, 
i bought my portapack h2 with hackrf from aliexpress.com.
it arrived with mayhem v1.4.0
and it was working perfectly, until i was downloading the new firmware from github, and installed mayhem V1.4.2

after that i just have a black screen, it doesn't boot up att all, i have tryed several times reflashing it, but the same issue accure. 
i have a thought of maybe the chipset is different on my portapack since it has been bought from aliexpress.... i read some reports something about that on github, 

i have tried finding the modified firmware from 
link 1: http://www.taozhengbo.com/mayhem_v1.3.0_FIRMWARE_8731.zip
link 2 : https://www.mmdvm.club/usr/uploads/2020/12/2458645348.zip
but all of the chinese sites are down. 

could anybody help me getting my portapack up and running.
maybe one of you guys have downloaded the modified firmware 1.4.0 or v1.4.2 
and could share the firmware here to me?

best regards stein 
 
[image]

[image]
in

Stein3247, 2021-12-04:
I can see that i have the same chip on my portapack h2 as reported i issue #264
There they had to install a modified firmware

Could anybody send me a copy of the modified firmware if any body have it on the pc?

[image]

sck-nogas, 2021-12-04:
Hmmm, if you have a WM8731, then [the commented out code on line 184 in "firmware/application/portapack.cpp" is the issue](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L184)..

I created a kludge of a [pull request](https://github.com/eried/portapack-mayhem/pull/439) to let us build a version of the firmware for the older audio chip, since it currently can't be detected in code. 

Or you can edit your code and comment out the "[model = PortaPackModel::R2_20170522](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L187)" line and uncomment the "[model = PortaPackModel::R1_20150901](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L184)" line.

Stein3247, 2021-12-04:
Ok, so its actually possible to fix this? 
Could you please help me maybe edit it for me, or tell me what to do step by step, because this i all new to me about how to edit
 "model = PortaPackModel::R2_20170522" line and uncomment the "model = PortaPackModel::R1_20150901" line.

Thanks

sck-nogas, 2021-12-04:
First off. If you haven't ever [compiled your own firmware, then there is a BIT of tweaking involved](https://github.com/eried/portapack-mayhem/wiki/Compile-firmware). But, I have [uploaded my compiled firmware of 1.4.2](https://nogas.org/Portapack-Mayhem/portapack-h1_h2-mayhem_R1_20150901.bin) if you want to try it...

But, you will have to do a [DFU to get your hackrf to accept new code](https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting). Then you can go through the [normal update procedure](https://github.com/eried/portapack-mayhem/wiki/Update-firmware).

Stein3247, 2021-12-04:
> First off. If you haven't ever [compiled your own firmware, then there is a BIT of tweaking involved](https://github.com/eried/portapack-mayhem/wiki/Compile-firmware). Then you can edit the source code to fix this issue (for yourself only)
> 
> But if you want the quicker path, I have [uploaded my compiled firmware of 1.4.2](https://nogas.org/Portapack-Mayhem/portapack-h1_h2-mayhem_R1_20150901.bin) if you want to try it...
> 
> But, you will have to do a [DFU to get your hackrf to accept new code](https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting). Then you can go through the [normal update procedure](https://github.com/eried/portapack-mayhem/wiki/Update-firmware).

thank you so very much sir 👍 💯  i will try this straight away 💨  il hope this fixes the issue that it dont boot up 😄

Stein3247, 2021-12-05:
> First off. If you haven't ever [compiled your own firmware, then there is a BIT of tweaking involved](https://github.com/eried/portapack-mayhem/wiki/Compile-firmware). Then you can edit the source [...]

Hi! I tried the file you uploaded (https://nogas.org/Portapack-Mayhem/portapack-h1_h2-mayhem_R1_20150901.bin), but i have the same issue, the portapack is still not turning on, its still a black screen. Do you have any ideas of what to try out?

sck-nogas, 2021-12-05:
Well, you can DFU back to a working HackRF... 

But, there is something "strange" about your Portapack. 

I have the original Portapack H1 (20140826) version, and when I compile 1.4.2 the only issue I have is the misconfiguration of the audio chip from "PortaPackModel::R2_20170522" back to the old "PortaPackModel::R1_20150901" 

But your H2 board has the wm8731 audio chip in it, but H2 versions should have the ak4951 audio chip. 

So, there is likely something going on with the initialization. The setting toggles between different versions of...

- audio chip
- [cpld](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L203)
- [backlight](https://github.com/eried/portapack-mayhem/blob/e2a2f7a448bbc16cf7241a4141fe9d0d5414355a/firmware/application/portapack.cpp#L209)
God knows what else is changed with those settings. 
-

aesaganda, 2021-12-05:
Similar issue happened to me #440 . Maybe leftover data at flash chip prevents Portapack from booting. After erasing flash chip and following DFU update procedure, my Portapack started to work as usual.

Stein3247, 2021-12-05:
> Similar issue happened to me #440 . Maybe leftover data at flash chip prevents Portapack from booting. After erasing flash chip and following DFU update procedure, my Portapack started to work as usual.

I just read tru #440 and Yeah! Thats just the same problem as mine! 
The only problem is that i am not god att coding. 
Could you please tell me step by step of what to do 😃 i would be greatfull happy, because i have been pulling my hair of now the last days😅

aesaganda, 2021-12-05:
Yes, I have bought mine from Aliexpress as well. I am going to try to explain what I did in order to recover the device.

First, you need a PC installed Ubuntu or Debian. A virtual machine will work as well.

Than you should install HackRF-host tools from [https://github.com/greatscottgadgets/hackrf/tree/master/host](url). The guide explains how you can compile the tools in detail. After compiling the tools I recommend restarting your PC (or virtual machine).

After that step you need to detach Portapack extension from HackRF board and plug only the HackRF board to PC.

At that point you need to put HackRF to DFU mode. In order to do that you should press the top two buttons at the same time, than release the button near the edge and finally release other button. If you are using a VM, you also need to pass USB from host to guest.

To check if the device is in DFU mode simply type to terminal `lsusb | grep NXP`.
It should return something like this: `Bus 001 Device 070: ID 1fc9:000c NXP Semiconductors `

Now it is time to flash the firmware to HackRF. Right click at the downloaded firmware folder _(mayhem_v1.4.2_FIRMWARE)_ than press _Open in Terminal_.

At that window type `sudo apt install dfu-util`. That command installs the software needed to flash firmware to HackRf. After installation type `dfu-util -D hackrf_one_usb.dfu`. At that point LEDs on top of your HackRF should light brighter than before. That points it has successfully flashed firmware to device. 

Than type `hackrf_spiflash -c` to terminal. That command erases flash chip and ensures no leftover data remained on it. 

Finally typing `hackrf_spiflash -w portapack-h1_h2-mayhem.bin` flashes the firmware to your newly cleaned chip. If you want to be sure if pc recognized it as a HackRF device you should type `lsusb | grep Open`. 

The output should be like this: `Bus 001 Device 071: ID 1d50:6089 OpenMoko, Inc. ` . If you see that message the process is successful and you can plug your HackRF board to Portapack and use it as usual.

If you have questions about DFU recovery process, you should watch [https://www.youtube.com/watch?v=_zx4ZvurgOs](url).

I hope these steps will fix your problem.

Stein3247, 2021-12-06:
> Than you should install HackRF-host tools from https://github.com/greatscottgadgets/hackrf/tree/master/host.

Wow what a community😃👌🏻 thanks for all the help so far, i will report back as soon i have tested this. At the moment i'am stuck on 
"install HackRF-host tools from https://github.com/greatscottgadgets/hackrf/tree/master/host."

I have a windows and a mac, to try this on, i dont have a linux to install hackrf host tool on, 
This was not as easy as i was hoping for. I havent done this as much as you guys. So sorry for the late reporting back. I thing iam gonns try to install the hackrf host tool on windows, do you know if it is possible to install the hackrf host tool on the macbook?

aesaganda, 2021-12-06:
Actually, you might compile the host-tools for Mac as well but you will probably run up against dependency issues. Since I haven't got a Macbook, I haven't experimented with compilation process, so I can't help you much. And Windows way is usually more grueling than Linux, if you could not manage to compile it on Windows, virtual machines are your friend.

AmiiiiKo, 2021-12-09:
I had the same problem, I even have the same case.
I couldn't even get DFU to connect.

I ran the [portapack firmware tool](https://github.com/sharebrained/portapack-hackrf/wiki/Updating-Firmware-Windows) designed for the original portapack, then the [Mayhem update](https://github.com/eried/portapack-mayhem/releases/tag/v1.4.2) and it worked as it was meant to.

Stein3247, 2021-12-09:
> I had the same problem, I even have the same housing.
> 
> I couldn't even get DFU to connect.
> 
> 
> 
> I ran the [portapack firmware tool](https://github.com/sharebrained/portapack-hackrf/wiki/Updating-Firmware-Windows) designed for the original portapack, then the [Mayhem update](https://github.com/eried/portapack-mayhem/releases/tag/v1.4.2) and it worked as it was meant to.

Yeah i have tried that to, without any luck. It just sems like it forgets everything after the reboot, afte i have flashed it with the new firmware. 
I think it has something to do what Aesaganda says, it has to have some files in the flas memory that dont get deleted. 

And i am not getting this hackrf host tool up and running on my mac or windows. 

Its realy a pain in te ass, could anybody tell me maybe a how to do step by step building a havkrf host tool on a mac? If it is possible

sck-nogas, 2021-12-09:
My environment on a my Mac...

1. Install [Xcode](https://apps.apple.com/us/app/xcode/id497799835)
2. Install [Gnu ARM Embeded Toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) into your ~/bin/directory (or wherever you put it make sure it's in your path)
3. Install [MacPorts](https://www.macports.org/install.php)
4. Open Terminal Window and run the following commands ( the stuff after # are comments)

```
sudo port install cmake git       # Install the git and cmake utilities in your environment
cd ~               # cd into your home directory (just to make sure you are in the right spot)
mkdir src       # make a src directory in your home dir (if you don't already have one)
cd src            # cd into your src dir
git clone https://github.com/eried/portapack-mayhem     # Git clone a copy of the portapack-mayhem source code
cd portapack-mayhem      # cd into that new source directory
git submodule update --init --recursive    # initialize and update the submodules needed
export PATH=~/bin/gcc-arm-none-eabi-10.3-2021.10/bin:${PATH}  # set your path to the compiler you installed in step #2
cd firmware      # cd into the firmware directory
rm -rf build      # optional on the first run (in case you are recompiling) to clean up the previous compiles
mkdir build      # create a build directory (to build in)
cd build           # cd into that build directory
cmake -B./ -S../../    # call cmake to do it's magic
make firmware        # call make to build the firmware
hackrf_spiflash -w firmware/portapack-h1_h2-mayhem.bin       # once the firmware compiles, use this to install your firmware
```
Now, that's the "steps" I have scribbled down to do a build on MacOS, but if you are not familiar with git/make/shell commands, then this can feel very overwhelming.

Stein3247, 2021-12-12:
It finally worked😃😃 i followed all of your steps! Thank you everyone! Its up and running again! What a wonderfull comunity!!! 
I had to type in Hackrf_spiflash -c 
So i Deleted everything before on the spi flash. 

Thank to everyone who helped me! 

I stugled many days to make hackrf to speak with the mac. And finaly

ArjanOnwezen, 2021-12-12:
Issue solved.

closed 2021-12-12 by ArjanOnwezen