portapack-mayhem/mayhem-firmware #453: Battery percentage on top of the screen along with the other icons...

saormart, 2021-12-21:
Would be nice to have a battery percentage info on top of the screen along with other little icons....
I know that the portapack have those 4 leds that represents the battery percentage, but some portapack enclosures covers those leds and it's not possible to know when you must charge your battery!

I'm new in sw. dev. and I'm trying to figure out how those 4 leds works to try to convert it into a piece of code to show in the screen, so I could contribute to my own request...
If some one have any idea I'm open to accept any help!
Thanks!

ArjanOnwezen, 2021-12-21:
@saormart 
It's not possible to measure battery power without hardware mod afaik
You can add an external battery indicator.

saormart, 2021-12-21:
> @saormart It's not possible to measure battery power without hardware mod afaik You can add an external battery indicator.
Got it! So those 4 leds are probably controlled by it's own IC that charges them as I suspected...
I'll think on some other solution!
Thank you!

eried, 2021-12-22:
I mean if the mod is simple enough, i.e. a voltage divider to a pin xx, it would be a nice addition. But otherwise it is a stretch of a "suggestion" :)

saormart, 2021-12-22:
> I mean if the mod is simple enough, i.e. a voltage divider to a pin xx, it would be a nice addition. But otherwise it is a stretch of a "suggestion" :)

I'll check it out, and I may create a simple voltage divider that we can connect it to a pin xx ...
If so, I'll post back here the mod, and I may need some help to implement it to show on the screen!

saormart, 2021-12-22:
> I mean if the mod is simple enough, i.e. a voltage divider to a pin xx, it would be a nice addition. But otherwise it is a stretch of a "suggestion" :)

I just figure out that my portapack uses the IP5306 is a multi-function power management SOC with an integrated boost converter, lithium battery charge management, and battery power indication, providing a complete power solution for mobile power supplies.
This IC is responsible for the 4 leds battery status!

Here's the datasheet link: [https://datasheetspdf.com/pdf-file/1139129/Injoinic/IP5306/1](https://datasheetspdf.com/pdf-file/1139129/Injoinic/IP5306/1)
I'll analyze a possible way to properly come up with a simple solution... we may need more I/O pins to implement it!

saormart, 2021-12-22:
@eried, based on the information I mentioned above about the IC IP5306, should I reopen this ticket?
We will need 3 I/O pins on the hackrf in order to get the battery percentage (25% 50% 75% 100%) ...
I have good experience programing Arduino/raspberry pi .... I may able to contribute to this but I may need some help!

ArjanOnwezen, 2021-12-22:
@saormart If you find a good solution, why not.

saormart, 2021-12-22:
Now I'm a little confused... according to the schematic "portapack_h1.pdf" in the repository, it's a bit different from the pcb I have....
We may need to know how many versions of these boards are out there, and if they show the battery status similar the one I have based on the IC "IP5306"

Anyway I'm reopening this ticket because I believe that our community can contribute to this!

saormart, 2021-12-22:
According to the datasheet, the IP5306 have a I2C interface (based on the model), but the schematics doesn't seems to have the I2C pins or.... maybe a pair of the LED's can be used as I2C... I'm trying to figure this out... If this IC have a I2C interface, we may can use it by disable the leds, and wire the i2c interface and this code maybe helpful:
`int8_t getBatteryLevel()
{
  Wire.beginTransmission(0x75);
  Wire.write(0x78);
  if (Wire.endTransmission(false) == 0
   && Wire.requestFrom(0x75, 1)) {
    switch (Wire.read() & 0xF0) {
    case 0xE0: return 25; //25%
    case 0xC0: return 50; //50%
    case 0x80: return 75; //75%
    case 0x00: return 100; //100%
    default: return 0;
    }
  }
  return -1; //Fail to read....
}`

saormart, 2021-12-22:
I was happy that this IC "have" a I2C interface, but I was wrong.
For some reason seems the company that built it, they did another version of it and for some reason they keep the same part number that's making alot of people confuse about this chip...
Here's what I found: the I2C interface register! (probably for the newer chip version...)
Here's the datasheet including the I2C registers interface: [https://sharvielectronics.com/wp-content/uploads/2021/07/IP5306-I2C-registers.pdf](https://sharvielectronics.com/wp-content/uploads/2021/07/IP5306-I2C-registers.pdf)
We may still be able to use the led's output to get the battery % ....

saormart, 2021-12-22:
Update: I found a Lithium charger with a similar circuit supposedly using the old IP5306 (like the one we probably all have on our portapack), and I ordered it to perform some tests and do a possible solution for this case....
It may take a while to arrive due it's coming from China.
I'll post the updates after I get it in hands, and do some tests...

jamesshao8, 2021-12-23:
Good idea. Already did that on H3.
[image]

saormart, 2021-12-23:
> Good idea. Already did that on H3. [image]

Hi there, could you please provide more details how you did it?
Thank you!

jamesshao8, 2021-12-23:
Redesign the pcb and write the driver code. I also added a lot other chips.

ArjanOnwezen, 2022-04-26:
Not possible with current hardware.

closed 2022-04-26 by ArjanOnwezen

AndKe, 2023-06-05:
If we have one spare AD input, the only required hardware needed is two resistors for a voltage divider - I'd gladly make instructions on how to do that - if the software end is supported (read out AD , and multiply with a number to get voltage)

xsesorigins, 2024-01-06:
Actually there is a battery indicator I figured it out when adding a speaker. There are 4 leds around the know. 25% 50% 75?% 100%

selevo, 2024-10-13:
in the 21st century, displaying a battery charge indicator on the LEDs that are located under the encoder wheel, while there is a display, is one of the most unfortunate solutions that one could come up with. Moreover, the light of these LEDs will interfere with watching the screen. and if the case is transparent, then it looks even worse, in the form of some kind of annoying illumination of the case. and flashing.

htotoo, 2024-10-13:
So, h3 has a voltage based percentage display, that is not the best one, since it can jump up and down a lot with load.
The h2 needs a huge hw mod to do that, that is not suggested.

The new h4 has this feature, and that has a specific ic for thet that is learning based on usage and gets better after every cycle.

Kingdamienjl, 2024-12-07:
So what do I do to complete a cycle for the soc to improve? Let it die completely twice?