portapack-mayhem/mayhem-firmware #454: Add more detailed maps

ArjanOnwezen, 2021-12-30:
Although working fine, currently the maps aren't that detailed.

It would be good to use for example the open source maps of MapsForge (https://github.com/mapsforge/mapsforge), which provide a better level of detail. There's already a C++ library. https://github.com/shentok/libqmapsforge

Not sure if the device is powerful enough to render those maps though.

eried, 2022-01-03:
The current size is the maximum the device can handle with the current way the map works (not tiles). I tried to compose a good combination between features like sea deep levels and labels for the one included (I think I even made a pool in a fb group with possible map styles)

Are you suggesting to change that map? (or to improve the size of the map we can display) because it would be interesting to see how mapsforge looks.

ArjanOnwezen, 2022-01-03:
Mapsforge has a way higher level of detail. Same as openstreetmap. See for example a screenshot of an app using MapsForge.
[image]

ArjanOnwezen, 2022-01-03:
Off course with the library mentioned it's possible to skip rendering certain objects.

eried, 2022-01-03:
yeah, this is similar to what I did, but you have to keep in mind the limit of 4 gb of the sd. We have an uncompressed bitmap now, and no tiles. To create the current one I used opensea+openstreetmaps combined:
[image]

All the tiles of a certain zoom level that fitted in a big bmp, plus some zoom out. The original old map was very rudimentary, my city just appeared as a smudge :P 

To improve the details I think it needs an architectural change on how the map display works:

1. We need tiles (not a giant bmp)
2. Zoom levels (easier with tiles)
3. Compression? (there will be a lot of empty tiles)

ArjanOnwezen, 2022-01-03:
Clear. Yeah the file from mapforge are per country, but extremely detailed. Obviously having the whole world on SD is impossible, but country level is doable.
So basically we need:
- standard solution
- not too big files (possibly having files per country/region)
- more detailed maps

[image]<https://user-images.githubusercontent.com/1091420/147935905-a68a29cc-1123-4ed2-8746-218c8dac5ee5.png>

TriniHam, 2022-03-16:
> yeah, this is similar to what I did, but you have to keep in mind the limit of 4 gb of the sd. We have an uncompressed bitmap now, and no tiles. To create the current one I used [...]

Okay I see what you you mean but can I custom upload a particular part of those detailed maps on my Portapack? if so how to do that and gain access to those file or create them myself?

eried, 2022-03-17:
You can create your own map following this: https://github.com/eried/portapack-mayhem/wiki/World-map-generation with tiles that you can get using something like https://www.allmapsoft.com/omm/ but there is no functionality to "zoom" or focus on a region, so the zoom level will be always the same

TriniHam, 2022-05-22:
> You can create your own map following this: https://github.com/eried/portapack-mayhem/wiki/World-map-generation with tiles that you can get using something like https://www.allmapsoft.com/omm/ but there is no functionality to "zoom" or focus on a region, so the zoom level will be always the same

Okay so I used the Offline Map Maker but how do I specify the px to be 32768 x 32768 px? 
the maximum I'm getting is 1024, do I have to resize in photoshop or something?

closed 2023-12-04 by eried