portapack-mayhem/mayhem-firmware #455: About portapack H2 + hackrf one updating 1.4.0 black screen

Bg7ena, 2021-12-31:
About portapack H2 + hackrf one updating 1.4.0 black screen

Hello, sir. I am a new user of portapack H2 + hackrf one. Excuse me, I'm in your GitHub The firmware of 1.4.2 downloaded on the COM / eried page is brushed into the new device I purchased. The display screen does not show a black screen. What is the reason? Does the firmware file of portapack need to be modified before it can be used normally? Can you ask me how to modify it? Thank you very much, thank you! Here is a novice user of bg7ena! Finally, I wish you a happy new year in advance!

MacGnG, 2021-12-31:
Can you provide additional information? do any of the lights work? is the battery charged? have you unplugged it and then plugged it back in? was it working properly before you upgraded? did you try a different cable and the tried upgrading again? do you get any response from the hackrf_info command in the command line? does your computer show that the hackrf is connected to your computer via the usb cable?

Bg7ena, 2021-12-31:
First of all, thank you very much for your reply! The two boards are combined together, and then with a battery. The computer is connected to the PC through a USB cable. It works normally after receiving the goods. The working version is 1.4.0. It seems that this version of the portapack program has been edited, which is different from all versions of GitHub. The following is the edited firmware of 1.4.0 I received. After all, I am a beginner. Please see the difference between this version and GitHub. Thank you! I also wish you a happy new year in advance! This is bg7ena!

 

 

从QQ邮箱发来的超大附件

mayhem_v1.4.0_FIRMWARE_cpld.rar (3.69M, 无限期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?k=6d326237bd82fd9912da6b2211610b490f0301515657000014035106064c0d050e014f5651020f4b5b51530403030b5508005256377e390b584b0a525a3e4f5717064c076827703474652365723e5a1655564c455613395b&t=exs_ftn_download&code=92b77a9f

MacGnG, 2021-12-31:
If the version on this github is not correct.... maybe ask the person that provided you with 1.4.0 to provide you with 1.4.2?

Bg7ena, 2021-12-31:
I'm trying to figure out what files they've changed, so I can edit whatever version I want. They haven't done 1.4.2 yet.

eried, 2021-12-31:
There is some differences with some hardware, we still don't have a perfect solution so you should contact the seller for now 😊

Bg7ena, 2021-12-31:
Thank you for your reply! Mr ErwinRied! Can I send you the firmware the vendor sent me? After that, please help me analyze where the firmware was modified by the merchant. Then teach me how to install all firmware on my machine? Thank you very much! Here is the bg7ena! Also wish you a happy New Year! Good health!

saormart, 2021-12-31:
> Thank you for your reply! Mr ErwinRied! Can I send you the firmware the vendor sent me? After that, please help me analyze where the firmware was modified by the merchant. Then teach me how to [...]

I think you should contact your seller once he's responsible to maintain the supported firmware for your device!
Mine I got from Aliexpress, but the seller pointed out this repository that support the hardware he's selling, so he avoid future trouble!

ArjanOnwezen, 2022-01-01:
> Thank you for your reply! Mr ErwinRied! Can I send you the firmware the vendor sent me? After that, please help me analyze where the firmware was modified by the merchant. Then teach me how to [...]

This only works if you have the source code..
Please check withe the vendor.
We can't / won't reverse engineer the compiled firmware.

aesaganda, 2022-01-01:
You might want to look issues #440 and #445. Maybe your problem is caused by leftover data at flash chip. Similar case happened to my clone Portapack and I have managed to solve my problem by erasing all leftover data in it.

Bg7ena, 2022-01-01:
Thanks for your reply! All right, I'll check out the questions. I'll ask you later! Thank you very much! Happy New Year!

Bg7ena, 2022-01-01:
Thanks for your reply! I'll consult someone else!

doccodyblue, 2022-01-07:
Run into the same situation. Probably the same clone. @Bg7ena Any news on that? Probably not the right place to discuss here. went through the mentioned #440 and #445 already without any luck

Bg7ena, 2022-01-10:
Hello! Their previous post seems to have solved the problem. It is estimated that it is to empty the data in it and then write it again. This operation is a little difficult for me. So I didn't go on! I asked the seller for the firmware. Anyway, it can be used for the time being. I posted it before. You can find it if you need it.

------------------&nbsp;原始邮件&nbsp;------------------
发件人: ***@***.***&gt;; 
发送时间: 2022年1月8日(星期六) 上午6:08
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [eried/portapack-mayhem] About portapack H2 + hackrf one updating 1.4.0 black screen  (Issue #455)

***@***.***这方面有什么消息吗？可能不适合在这里讨论。#440和#445已经没有任何运气

—
直接回复此邮件，在 GitHub 上查看或者取消订阅.
GitHub Mobile 上的分流通知IOS 系统或机器人. 
你收到这个是因为你被提到了。留言 ID ***@***.***与&gt;

ArjanOnwezen, 2022-01-13:
@Bg7ena sended back his unit to seller.

closed 2022-01-13 by ArjanOnwezen