portapack-mayhem/mayhem-firmware #456: What to buy?

rd0fun, 2021-12-31:
Apologies in advance, this is not a "bug", but a request to prevent issues. I see there are hardware things to avoid to have a good experience. 

So, what specific hardware from what particular seller to you all suggest for a brand new user?

Thanks in advance.

BartenderWinery, 2021-12-31:
You shouldn't leave issues for requesting suggestions. 
Usually, more expensive hardware equals better performance.

jumbo77, 2022-01-01:
The main portapack-mayhem home page what is have the picture of portapack you can choose and have a link.

eried, 2022-01-03:
Yeah, that link was the one I bought years ago, but they changed the item with "that" incompatible audio chip few months ago. I changed the link now to the one with most sales in aliexpress, hopefully they will handle any problem better (i.e. put some resources doing a PR, since they probably have many test units).

ArjanOnwezen, 2022-04-06:
For advice, see link on main page

closed 2022-04-06 by ArjanOnwezen