portapack-mayhem/mayhem-firmware #462: Scanner only outputs sound on 1st frequency hit

bluegizmo83, 2022-01-04:
**Describe the bug**
The scanner app only outputs sound the first time a frequency is hit. After that, every time it stops on a frequency where there is a signal (I verified there really is a signal with a second sdr), it does not output any sound. I've tried playing around with the squelch but that made no difference.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to scanner app
2. Start any scan

**Expected behavior**
The scanner should output sound every time it stops on a frequency with a signal.

**Affected versions**
1.4.2 (haven't tried previous versions)

anomalyLuna, 2022-01-05:
If you change the volume once you land on a frequency it does output sound. This is still a bug as you have to move the volume up or down every time you land on a new scan frequency in order for sound to be heard.

bluegizmo83, 2022-01-05:
> If you change the volume once you land on a frequency it does output sound. This is still a bug as you have to move the volume up or down every time you land on a new scan frequency in order for sound to be heard.

Ah ok, thanks!

ArjanOnwezen, 2022-01-06:
I can't replicate this behaviour. It scan from frequency to fequency and I can hear output through the headphone on a portapack H1.

bluegizmo83, 2022-01-06:
> I can't replicate this behaviour. It scan from frequency to frequency and I can hear output through the headphone on a portapack H1. Do you have this on the headphone or do you have a built-in speaker?

Hmm, must be a issue that only appears on the H2, possibly only on the versions with the WM8731S audio chip...

I'll play around with modifying the firmware to see if I can come up with a fix.

euquiq, 2022-01-06:
I can´t replicate this behavior, also. H1 here ... I can use the scanner just fine. 

BUT THEN; I am using my "original" scanner PR. Maybe you can try to download an earlier mayhem firmware and try it ? Like:

https://github.com/eried/portapack-mayhem/releases/tag/v1.2.3

So we can determine if this is either an WM8731S H2 specific problem, OR there is new code added later, somewhere, introducing problems ? 

Thx!

ArjanOnwezen, 2022-01-06:
@euquiq Perhaps a post on the Discord channel.

bluegizmo83, 2022-01-06:
> I can´t replicate this behavior, also. H1 here ... I can use the scanner just fine.
>
> BUT THEN; I am using my "original" scanner PR. Maybe you can try to download an earlier mayhem firmware and try [...]

Ok I tried version 1.2.3 but I get no audio at all with that version, and not just in the scanner. I'm guessing maybe the WM8731S audio chip wasn't implemented yet in that version or something?

bluegizmo83, 2022-01-06:
Ok I've found a fix for this issue. This does not fix whatever the root cause of the problem is, so I guess you could call it a bandaid fix, but it works and its simple.

I've never contributed to someone else's GitHub project before, so I'm not sure how to go about pushing code to this repository like other people do, so I'm just going to post the fix here and hopefully someone who knows how can implement it in a future build.

The fix is one single line of code added to the `ui_scanner.cpp` right after `audio::output::start();` as seen here:

**ORIGINAL CODE BLOCK**
```
void ScannerView::scan_pause() {
	if (scan_thread->is_scanning()) {
		scan_thread->set_freq_lock(0); 		//Reset the scanner lock (because user paused, or MAX_FREQ_LOCK reached) for next freq scan	
		scan_thread->set_scanning(false); // WE STOP SCANNING
		audio::output::start();
	}
}
```

**UPDATED CODE BLOCK**
```
void ScannerView::scan_pause() {
	if (scan_thread->is_scanning()) {
		scan_thread->set_freq_lock(0); 		//Reset the scanner lock (because user paused, or MAX_FREQ_LOCK reached) for next freq scan	
		scan_thread->set_scanning(false); // WE STOP SCANNING
		audio::output::start();
		on_headphone_volume_changed(field_volume.value());
	}
}
```

ArjanOnwezen, 2022-01-06:
Issue might be in https://raw.githubusercontent.com/eried/portapack-mayhem/firmware/common/wm8731.hpp
It looks like the unmute set speaker volume to -60dB regardless of previous level.

bluegizmo83, 2022-01-06:
> Issue might be in https://github.com/eried/portapack-mayhem/blob/next/firmware/common/wm8731.hpp It looks like the speaker_enable / unmute set speaker volume to -60dB regardless of previous level.

Yeah that definitely looks like it could be the root cause. I know with my headphones plugged into the Portapack I can't hear it hardly at all when the volume is set below about 50%. Not sure what that would be in Decibels, but if -60dB is less than that, it could explain why I thought it wasn't putting out any sound... Just from a quick glance at that code though, I'm not sure how to go about fixing that hard coded -60dB level to instead retain the current volume level on unmute though.

icepicknz, 2022-01-13:
I received my H2 the other day and have been trying to work this out myself too.
Device came with 1.4.0 and have now upgraded to 1.4.3 and same issue

Basically I have to use volume control to enable audio output when a channel is found.
Another weird thing is the audio is distorted like chipmunks or something.
However if I go to "Audio" down the bottom in scanner, I can hear it properly

ArjanOnwezen, 2022-01-28:
@icepicknz / @bluegizmo83   Could you test PR #486 No clue if it works, since I can't test this, since I only have the AK one.

ArjanOnwezen, 2022-04-08:
Solved with solution suggested by @bluegizmo83 in #555 . Please try the first nightly

closed 2022-04-08 by ArjanOnwezen