portapack-mayhem/mayhem-firmware #464: Problem upgrading the firmware

LoekVeeger, 2022-01-07:
Hi there,
I have a brand new portapack H2, with 1.3.1 mayhem firmware. It works ok stand alone and if I connect it to my laptop, I can work with SDRangel like a charm. To connect I have to choose in Angel for the device Soapy Hackrf one,.... I have installed Pothos.
If I exit SDRangel, I want to upgrade to 1.4. I downloaded the firmware and the 'copy-'software.
I unzipped everything to a folder on my pc. The copy stuf goes into a flash disk, and into the portapack. 
Than I want to make the firmware update. I start the .bat file according to the documentation. It states that I should connect the portapck to the pc and put it in the Hackrf mode. Ok, I did that and the blue screen appears on the portapack. Like using it with Angel.
I hit the spacebar and the .bat file states that he can not find a hackrf one. I re downloaded and redid the whole procedure, but no recognition of the hackrf one.
After taht I started sdrangel again,.....no problem, everything works ok. Back to the bat file,....no hackrf found again.
than the special procedure,.......dru,..etc. No hackrf found. 
It doesn't work.
Never did it before with the brandnew device. Any idea how to procede.

Thanks for any advise,

Loek Veeger, PE0MJX (ham callsign)

anubus, 2022-01-14:
Make sure you are carefully following the instructions and then the trouble instructions as well (https://github.com/eried/portapack-mayhem/wiki/Update-firmware-troubleshooting). Watch the video, particularly the first. It seems you may have to update your windows driver with the version in the firmware version download.   Also, I recommend joining the Havoc / Mayhem Facebook page for these kinds of questions.

ArjanOnwezen, 2022-03-26:
No response from reporter @LoekVeeger, closing due to inactivity

closed 2022-03-26 by ArjanOnwezen