portapack-mayhem/mayhem-firmware #465: Add 'enable ALC' in Microphone app where AK4951 codec is available

gerryk, 2022-01-09:
Where a microphone has poor dynamic range, over-modulation in AM and SSB modes can occur, resulting in harmonic distortion, as seen in this comment... https://github.com/eried/portapack-mayhem/pull/457#issuecomment-1008278584

Enabling ALC (p54 of datasheet) in the AK4951 audio codec may be a mitigation.

Additional strategies may be:
Microphone Sensitivity Correction (p46 of datasheet)
Automatic Wind Noise Reduction Filter (p47 of datasheet)
[12-AK4951EN.pdf](https://github.com/eried/portapack-mayhem/files/7835114/12-AK4951EN.pdf)

Brumi-2021, 2022-02-06:
Hi , @gerryk  , I tried to contact you directly though discord , but so far no reply from you...

 I have been  checking  and investigating locally about i t: Automatic volume Mic  Level Control, (ALC )  on AK Audio codec IC  , 
 and now I  have a clear idea about how to activate it and the initial  good  audio settings . 
It improves the audio compression , when we are at some distance to the MIC >10 cms , making it more sensible capturing other room audio voices or sounds. It reduces de audito saturation risk when getting closer to the mic   , but it does not solve completely when we are <2cms from the MIC or speaking louder . (that is why , I also suggest to add version (4) ALC -low ,  see below details) 

So far,  my current idea is to add  the following menu options (related to new ALC  AK4951 feature)  :
1. ALC:OFF  ( exactly same settings as now, NO CHANGE AT ALL  , I hope no confuse with ALC ;OFF , or it can be changed... 
2. ALC:HIGH  ( increased MIC pre-amp gain =current +9dB's   + ALC high  ) ,
3. ALC:MID    (  MIC pre-amp gain same us current + ALC mid ) 
4. ALC:LOW  (reducing MIC gain =current -6 dB's (ex.)   + ALC low ) ,

We need to discuss , where to add those 4 options , ONLY when detected AK ,.
 my current proposal is to add in option (1)  : 
(1-) Inside the MIC App.  , (we have space , in the first line ,  after MIC GAIN : , in the right part ) .
     (at the moment I will  try to test that one locally in next week/s, because it will be easy in the evaluation/testing   ) 

But if you want , we can also consider to  move into other more hidden specific AUDIO settings area , 
(2-) OPTION menu --> AUDIO SETTINGS ,  (maybe here is more simple way to add it , with persistent memory)

But in any case,  of course, 
i-) In oder to avoid confusions / doubts / --> WM8731 users should not see those options, (should be hidden to them) .   
        (we ONLY need to open to the GUI ,  when detecting properly ,  AK IC ) 
ii-)we should guarantee ,   NO side effects , to existing good part of WM8731 .

Therefore , we wiil  need deep test in both audio codec versions , to guarantee no side effects to the WM8731 
(because I only have H1  + AK version)
**_To be continued .... _**

Brumi-2021, 2022-02-13:
Hi ,  @gerryk (and all) , ALC is already working quite well ! 
But  I want also  to try to activate and combine it with  the Wind Noise Filter Reduction.   And, later ,  make general register fine tuning with different mics . And overall check carefully no side effects . 
Next week probably  I will upload some advance binary  test version in the discord test-drive.

Brumi-2021, 2022-03-04:
Hi  @gerryk and all friends , 

 I will upload today , that beta testing FW of that  ALC feature , in the usual HackRF Portapack Mayhem discord , test-drive channel , 

Later, I will send related PR , (but we will check also your feedback ) 
This , PR , tries to experiment and  minimize the Mic App overmodulation and spectrum harmonics generation when speaking too loud or too close to the mic (< 2 cms), 
[image]

In this new FW test, we are checking initially about which audio codec do we have in our portapack ,  if it is detected  WM8731 IC  , the menu does not show that ALC setting , (and the menu is exactly same as before ) 	

[image]

But  if  not detected audio codec WM8731 , we are assuming to have the other audio coded IC : AK4951 and therefore ,  we can activate all that Programmable digital blocks , as , ALC option settings in the GUI menu. 

The initial screen of the  Mic  Auto Level volume Control (ALC) is the following one , 

[image]

and you can select different  options of the Mic Auto Level volume Control.
I also combined it  with different digital filter  settings ,  (like example : ) 

[image]

[image]

Offering to the user to activate  the AK4951  Digital Automatic Mic  Level Control (ALC)  with several different settings to compress the audio signal dymanic margin and avoid that overmodulation, 

My initial expectation of that ALC function, was that it  will make a big dynamic control margin compression  , enabling to talk  from 1m or 2m distance to the mic , or also  touching the mic with the lips ,  but actually it does not cover such a big dynamic margin.   Therefore , we decided to open to the user , several max ALC recovery gain control settings , (+12 dB’s ,  +9dB’s, +6dB’s ,...0dB’s,  -3dB’s , -6dB’s , -9dB’s , -12dB’s.)    

Then user should set up his best   ALC settings  (we can check the mic sound Vumeter )
For instance , if we want ,to talk , like handsfree mode , and pick up all the voices and sounds inside our room (from 50 cms till  several meters distance from the mic,  we can select +12 dB’s ) , 
If we want to talk  around  15 to 20 cms , we can select  +0dB’s or -3dB’s depending on the loudness of our voice …
If we want to avoid saturation when shouting , or talking  touching the mic with the lips , you can reduce it til -9,  -12 dB’s

If you leave the default OFF settings , you will have exactly same Mic settings as in previous Mayhem FW versions <=1.4.3 ,  without activating any digital ALC block  of in the AK4951. 

When I said , “OFF - 20Khz” it just means that the audio base band output  from the IC 4951 has this initial max bandwith,  but obviously ,later , when it is post-processed and used for modulation , it is also reduced it . 

Please find attached the current combination settings behind the GUI menu ,  
I tried to use and combine several nice programmable digital features,  llike , ALC , Wind Nose Filter, LPF , and  boosting peaking Equalizer,, and also Pre-amp mic level. 

Ex.  +12dB-6Khz ,  means activated ALC with maximum gain of 12dB’s and with digital Low Pass Filter of 6kHz .

[image]

We will appreciate , to check specially that there is no side effect ,after using Mic App , going  to other App that uses also the  other audio IC blocks , like  (headphone , speaker, digital volume ,...) 

And  also confirm that with Portapack wiht WM8731 audiio codec  , that change is completelly transparent and no effect.

You can check the audio coded version in your Portapack ,going to “Debug App” and then select Peripherals . 

By the way ,  I already confirmed that before that PR ,
 the “Roger Beep” and the Tone Key (CTCSS) was already not working , from the version 1.4.0. onwards,  In next days , I will try also  to investigate it , and trying  to recover it .
 (first priority CTCSS) .  

Shortly i will release that related Mic -Auto Level Control  PR  proposal to integrate it in the “next” Mayhem release, if we do not find any bug or side effect. 

You can Test it , with  permanent PTT ,  (reducing radiated power , to the minimum that you need ) and then  reduce “LVL” till 0 or 2 , and then select  TX Activation to “AUTO” , and you will radiate all the room audio sounds to your next receiver ! 

I hope that you also like that additional Mic App feature !  Pls enjoy ! :-)

ArjanOnwezen, 2022-03-30:
Implemented by @Brumi-2021

closed 2022-03-30 by ArjanOnwezen