portapack-mayhem/mayhem-firmware #466: TX, but no modulation

DC2JR, 2022-01-10:
Hi,

I use the portapack with an original hackrf one and mayhem 1.4.2. All receive functions run well, on all TX function I get the RF out but only without any modulation. So using for example pcsag, Soundbrd, there is no modulation on the RF.
Using the hackrf in remote mode with a pc software , in TX I have modulation on the RF

gerryk, 2022-01-10:
Has this worked with any previous firmware versions?
Which PortaPack are you using?

gerryk, 2022-01-10:
This is strange, as the code for this stuff is pretty much PortaPack independent. The PortaPack is used to control the various apps, but once you start transmitting, it's all done on the HackRF. Have you tried one of the basic transmitters, such as Signal Gen? What are your Gain & Amp settings?

Brumi-2021, 2022-01-10:
HI , if you go to the MIC App , without pushing anything , when you talk to the connected MIC , the audio level vumeter -in the left part- is moving and following your voice ?

DC2JR, 2022-01-11:
@gerryk I tried the signal generator, I get RF out but no modulation, so it is the pure rf signal. as gain I have 47, as amp 0
@Brumi-2021 I tried out the mic app, I can see the audio level based on the speech

DC2JR, 2022-01-11:
@Brumi-2021 mmmh, I tried outgoing up and down kHz wise, I see based on my receiver, that the sending frequency is identical with the TX frequency of the hackrf, but I get no modulation. Are there any incompatibilities between firmware versions of the hackrf and mayhem? I use for hackrf the release 2021.03.1 with integrated CPLD.

gerryk, 2022-01-11:
The HackRF firmware blob is embedded in the Mayhem image. It was rebased 10 months ago so I am assuming it is the same version you mention.

HomicidalBrainiac, 2022-01-17:
I have this issue as well, but seems isolated to the replay function.

I recorded a few seconds on 462.624 MHz, then attempt to replay.  It’s nearly a blank carrier, comes off as heterodyning on another receiver even though no other carriers present.

I first thought to test this when replay attacks on my garage door opener failed.  Eventually I could get it work, but unreliability.  This is in total contrast with Havok firmware, just works every time.  Don’t get me wrong, Mayhem is a giant leap.

I did notice that when I used a transmitter in very close proximity (to where the front end is saturated a little) that replay does work, but parts of the replay has bits which are out of order and stuttering towards the end.

The soundboard app actually works quite well however.  You do hear a sorta buzzing and stuttering at the end of the transmission if looped.

Using the chinese H2 portapack avail on Amazon.

paddel57, 2022-01-17:
Now tried it also with 1.4.3 , no change of the issue

Brumi-2021, 2022-01-17:
Sorry to jump . 

I think that your problem description is not related with my below comments , but just in case ) 

In current FW version , for  correct replay application , you should  ONLY use capture app selecting BW <=600 Khz . (500Khz recommended for majority  micro SD cards compatibility)

 600khz may need fast micro SD Card , and with not so high random latency . (In the GUI , those bandwidth capture appears with the  Normal REC icon Background colour)

Above >600Khz  (with YELLOW REC button background Icon), in current  version , the recorded file is  “decimated” , and it has not recorded all  full original samples content and therefore  it is NOT  useful for the Replay App , just useful to check the spectrum image ). Anyway when replay  those captured files , the replay time will be shorter than real . but I got also modulation.

 If that is confusing you , pls let us know and we will block those higher Bandwith options (>600Khz).

Anyway I replayed several recorded files of 1Mhz ,... , and I always got correct modulation , but the recorded voice speed plays accelerated . (Then not useful for replay attack) 

in future firmware versions , we will try to  increase a correct recording .C16 files in 750Khz (and 1 MHz)  (beyond that , based on Portapack designers , our Portapack HackRF hardware system , will not be able to manage ) 

hopefully that limitation is not producing your no modulation  problem , but I just wanted to clarify it.

HomicidalBrainiac, 2022-01-17:
I have been recording 500 KHz and slower, even as low as 50 KHz and no dice.  I am now trying to troubleshoot using a spectrum analyzer.  

The spectral output so far is correct for the soundboard app but not replay. Im going to try reverting to havok and see if same issue.

Brumi-2021, 2022-01-18:
Ok, in the capture app, the bandwidth of 500 khz or below it has not been changed  l just added 600 khz and higher than that . Therefore 500khz capturing should work well as usual. When you compare it with havoc , pls make sure to compare it with similar radiated power  settings Amp, ...

ArjanOnwezen, 2022-01-28:
@DC2JR this looks like a HW issue (sd card). Do you still experience these issues?

DC2JR, 2022-01-28:
Yes I still have this issue and stopped quite frustrated the examination. I used also different SD Cards, but these should not the issue, when using the signal generator using as modulation a sine with 1000Hz in FM or AM and the result on the scope is only the pure rf signal

100monkeys, 2022-03-20:
Same here, No modulation at all, just carrier.
I use the same combination with original Hackrf one and a Portapack clone. With Mayhem set to hackrf i can get modulated signals with SDRangel so the Hack rf seems fine. Also tried older version Mayhem and Havoc. These can't modulate either. I can change the signal levels and the +14 db amp. Seems the Portapack clone has a defect? Shared frustration @ DC2JR ;)

100monkeys, 2022-03-20:
Shared frustration @DC2JR ;)
Same here, No modulation at all, just carrier.
I use the same combination with original Hackrf one and a Portapack clone. With Mayhem set to hackrf i can get modulated signals with SDRangel so the Hack rf seems fine. Also tried older version Mayhem and Havoc. These can't modulate either. I can change the signal levels and the +14 db amp. Seems the Portapack clone has a defect?

Brumi-2021, 2022-03-22:
Hello, in case that it may help you, let me share a new possible solution line ! 

Recently I was talking to Sharebrained -excellent teacher  and HackRF Portapack gurú . And he told me that ,  recently he has been contacted by **PortaPack customers who have had problems with transmit features, and manually flashing the HackRF CPLD fixed the problem.** 

In case that your CPLD IC is the same version as original , it may have some strange corruption inside. 

He suggested us , 
This is the guide I suggest for people having CPLD troubles:
https://hackrf.readthedocs.io/en/latest/updating_firmware.html#updating-the-cpld

if it works , pls let us know to share that know how to everybody l  and I will let him know about it. 🤞

100monkeys, 2022-03-23:
@Brumi-2021 
Thanks for the quick reply!

It works !
I unplugged the PP form the hackrf and then checked the firmware: 2021.03.1 
I did a reflash to make sure anyway.
I then did the "Updating the CPLD", got the 3 blinking leds and after reset all  TX signals now seem to modulate ok. 
Only missing the "rogerbeep" and the  "noise" in generator
Glad this finally works. I must admit i do not really understand the "cpld" thing but thanks a lot for the tip !
Also, i'm a noob in linux still but i managed to get it right.

Brumi-2021, 2022-03-23:
Hello @100monkeys , very glad to hear that you could solve the TX Modulation problem 👏👏👏👏👍

Be ready to be the teacher to our colleague @DC2JR , hopefully it will also work for him  , and we will celebrate it ! As soon as he also can confirm that solution we will close that issue ! 

Probably all of you bought it from the same wrong CPLD recording flash  batch . 

I am also newbie in Linux and Mayhem programming , but learning day by day 👍

Regarding the missing “roger beep” and the “key tones CTCSS features from current Mayhem Miac App FW , it is not your set problem, nor CPLD , It is just a bug , that I will try  to fix soon ! 

Thanks for your confirmation ,and  I will also share it and thanks to Sharebrained, who gave me that nice clue !

DC2JR, 2022-03-25:
Hi

I also tested and was successful!!!!!!!

73 de Klaus DC2JR

ArjanOnwezen, 2022-03-25:
Great news. Closing the issue..

closed 2022-03-25 by ArjanOnwezen

Brumi-2021, 2022-03-25:
Really great news ! 
Thanks for your confirmation . 
I am very happy Klaus , to be contributed  to fix your strange TX problem !    Thanks Sharebrained!!!
73

100monkeys, 2022-03-29:
Great @DC2JR

3qu1n0x, 2024-01-18:
Just a heads up in 2024: Had this exact issue with a China Clone and was able to fix it with @Brumi-2021 solution, so it's still relevant! Thank you and have a good year!