portapack-mayhem/mayhem-firmware #471: Can't decode the radiosonde QR code: it's too small!

periloso, 2022-01-17:
**Describe the bug**
As @UserAZERTY0 said in the pull request, the QR code is just too small, and I cannot decode it with my phone or my tablet, even by using macro mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Receive > Radiosonde > See QR code
2. Try to use your phone to decode the QR code
3. The phone won't decode it, as it's too tiny.

**Expected behavior**
The code should be decoded correctly.

**Affected versions**
Latest release (1.4.3)

**Additional**
As @UserAZERTY0 said, I propose to make it *at least* 2x bigger: right now, the feature isn't working and cannot be used; @ArjanOnwezen: the qr code doesn't need to be huge: you could use a shorter one like the one attached.
Thank you for your work!!
[image]

ArjanOnwezen, 2022-01-17:
Hi @periloso I've tried enlarging the QR code, but for some reason my phone doesn't pick that one up. So I resorted to the smaller one.

Possible because my phone camera see the QR bits not as a whole but as 4 different bits. One bit per pixel.
This might be because of the screen.

sliceratwork, 2022-01-17:
@ArjanOnwezen  is it hard to have a scale option for the QR code from the portapack UI?

Like turn the wheel and switch between x1 and x2.

ArjanOnwezen, 2022-01-17:
> @ArjanOnwezen is it hard to have a scale option for the QR code from the portapack UI?
> 
> Like turn the wheel and switch between x1 and x2.

It's not so hard. As I said my first version had it scaled, but it wasn't working.

ArjanOnwezen, 2022-01-17:
> **Describe the bug** As @UserAZERTY0 said in the pull request, the QR code is just too small, and I cannot decode it with my phone or my tablet, even by using macro mode.
>
> **To Reproduce** Steps to [...]

> **Describe the bug** As @UserAZERTY0 said in the pull request, the QR code is just too small, and I cannot decode it with my phone or my tablet, even by using macro mode.
>
> **To Reproduce** Steps to [...]

What phone are you using? I've tried this with several Xiaomi phones but they all work fine.
What sensor and/or QR code decoder do you use?

UserAZERTY0, 2022-01-17:
On my side tried with "QR Scanner" app on a BlackBerry Key2 LE.
Which usually works fine with QR codes but this one is too tiny for my phone that cannot find a correct focus - maybe a phone with macro or zoom mode can handle it but not mine.
A simpler way could be to propose a QR code with x2 and x3 options for testing and we could try it on our different configurations.

periloso, 2022-01-18:
*Devices tested*
I tried to use an honor 8, a lenovo p2, a huawei mediapad m3 lite 10" (tablet), and a microsoft lifecam hd 3000 (webcam); I wasn't able to decode the qr-code on any of them.

*Further testing done*
Back at home (with no radiosondes around), I tried to generate the qrcode, I took a screenshot, enlarged the image... and tried again. Using my monitor I was able to finally let my phone recognize the qrcode, but there was apparently no data inside it (no link, nothing...)

*Current ideas*
@ArjanOnwezen, I think that you're right to be concerned that the zoom won't work on all devices: you've already noticed this issue while writing your code and, btw, the display is certainly too small to add a lot of data;
@UserAZERTY0 proposed the zooming feature, but I believe it would require some more coding on somebody's end, and we're not even sure it would work on all devices.

*Source code review*
Since there was nothing inside the qrcode, I checked the source code and I noticed two things:
1. A lot of data is being added to the qrcode: that's why it's so dense, and so hard to decode;
2. The error tolerance algorithm used is "HIGH": the higher it is, the denser the qrcode gets. I believe that adding less data and setting the ECC to "NORMAL" would be fine.

*My proposal*
When I tried the feature, I just wanted to get closer to the radiosonde: if we need more details, I believe there's already the "log to sdcard" feature for that.
The compromise could be to use a simpler URL, a bigger qrcode, and a lower error tolerance setting... and getting a qrcode similar to the one I sent previously: to make it even safer, the URL can be formatted this way: https://maps.google.com/?q=41.91024,12.39557.

To @ArjanOnwezen and @UserAZERTY0: please let me know what you think, and thank you both for your time and efforts! 😊😊

ArjanOnwezen, 2022-01-18:
@periloso  
Let me clarify some things. 

**Coded QR text**
Coded QR text complies to geo URI scheme (RFC 5870), e.g. geo:37.786971,-122.399677. 
This should be recognised by all android phones and opens (user defined) Map app.
Also this is as short as it gets.

**Error tolerance**
Error tolerance has no effect on the QR size / density.
It only has an effect on the amount of data that can be stored.
Currently version 10 is used which allows for 174 alphanumeric characters using HIGH error correction level, as oppose to 395 characters using LOW error correction level. See also https://www.qrcode.com/en/about/version.html
The size of the QR code is determined by the amount of pixels used to draw the QR _bits_. Currently a pixel per bit is used in current version. During trails also 2x2 pixels for a QR bit were used, which results in a 4x bigger QR code.
However that didn't work well, due to the pixel density of the  PortaPack screen.

periloso, 2022-01-18:
@ArjanOnwezen, thank you for your clarification!
This is a photo I took with my phone of the screen of my portapack (it's a portapack H1, so I have a 2.4" screen).
As you can see, there are many more modules than I expected, which is why I suggested to also lower the correction level (more data with the same correction = bigger qrcode).

Since we're using the geo URI scheme (which is even better than what I proposed), the question is... do we really need to use version 10 to store about 25 alphanumeric characters? Can't we use version 1 or 2... with a bigger image and allow both users of portapack H1 and H2 to use the same feature?

[image]

UserAZERTY0, 2022-01-18:
Hi, just did a simple test:
- Opened the radiosonde app to see a QR code (even without radiosonde in the vicinity)
- Took a screenshot through the embedded feature of Portapack Mayhem
- Displayed it on my laptop via the SD card: it worked (screen is bigger), with various zoom levels it worked too (even with the biggest that causes some ugly visible blur, Android app seems not disturbed by that)

So I keep thinking a zoom option would fit the bill (or a x2-x3 image magnified by default)

periloso, 2022-01-18:
@ArjanOnwezen, don't you worry about this issue: I've already fixed it in my own local repo and, as soon as I finish the testing phase, I'll create a PR to close this ticket 😉
Thanks! ^^

ArjanOnwezen, 2022-01-19:
Hi @periloso . Great you already did a PR. 
However the PR isn't working on my phone.
You could make an option SHOW BIG QR CODE in setting (using persistent storage), allowing to show big QR code if enabled and else show the small one.
Also make sure everything is lined out well and also the QR text is shown correctly.

periloso, 2022-01-19:
Don't worry: I set the PR back as a draft until we have a release that works for everyone 😉

UserAZERTY0, 2022-04-02:
Was a bit disappointed to not see the magnified QR code in release 1.4.4
The current QR zoomed by 2 or 3 was working for me - the (little) current one is not (like for other users).
As such, I don't see the issue in not proposing something full working (QR zoomed 2-3x) if the current solution is not either... at least we would give additional coverage/possibilities.

jLynx, 2022-04-04:
+1 I am also having this issue

ArjanOnwezen, 2022-04-08:
All starting next nightly (or build yourself from Next branch) Option for bigger QR code has been added.

closed 2022-04-08 by ArjanOnwezen

UserAZERTY0, 2022-04-16:
Tested with FW 1.51: bigger QR code worked fine with me!

ArjanOnwezen, 2022-04-16:
Thanks for your feedback @UserAZERTY0 !