portapack-mayhem/mayhem-firmware #473: ADS-B not updating Altitude in Map

TekNicholas, 2022-01-19:
ADS-B does not update the altitude of aircraft live when in the map view. 

Lat and Lon are updated as well as the position of the aircraft on the map, but the altitude remains the same value it was when the map view was entered.  Altitude value works as it should in the previous Details page.

Workaround has been to exit map view and re-enter, but this is not very convenient. 

Bug was observed on v1.4.2 and recently confirmed also on v1.4.3.  

[image] [image]

zigad, 2022-01-23:
I was able to reproduce this issue. I fixed it and tried it on my HackRF H2 and can confirm it works now.

Easiest way to reproduce the issue is by going to ADS-B receiver and immediately click on the first flight and go to "See on map" straight away, you will see coordinates and altitude all 0 (expected since aircraft position was not yet received). If we wait a few seconds we will see the altitude stays at 0 but lat and lon will change correctly. With this fix #477 the altitude will also update with lat and lon as expected.

closed 2022-01-24 by ArjanOnwezen

Firecul, 2022-04-01:
Much better thank you