portapack-mayhem/mayhem-firmware #478: Pocsag

loewal, 2022-01-26:
Hi all,
I can't receive Pocsag anymore. The same antenna connected to my Mac gives many output.
On the Hackrf...nothing.

But it worked before.
ADSB works fine, so it's not an inputproblem.
I must say, that I had to buy a new Hackrf, because I connected the wrong side of the miniwhip (9v...) to my antenna.
After that my Hackrf got very hot. So, now I am using the new one.

I compiled and flashed the latest 1.4.3 firmware.
Any idea?

zigad, 2022-01-26:
Do you maybe know what type of POCSAF messages you should be receiving? Are they numeric, alphanumeric?
I noticed that "tone only" numeric messages are not decoded (they never were on my device), but alphanumeric messages are sometimes decoded for me.

loewal, 2022-01-26:
They are alphanumeric. Very many!

loewal, 2022-01-26:
I connected The HRF to a Linux-machine, running Ubuntu Impish and SDRangel with Pocsag pager demodulator.
Runs fine and shows many alphanumeric messages.

loewal, 2022-01-26:
I went back to 1.4.1 and noticed that now I can select polarity and baudrate.
With version 1.4.3 these options are disappeared!!!!!!!!!

zigad, 2022-01-26:
Hmm, I know there was a update for POCSAG decode. (Issue: #426) I think this was released in v1.4.2. The polarity and baud rate should now be detected automatically.

ArjanOnwezen, 2022-01-26:
@loewal What settings (baudrate / polarity were you using) and are messages showing up in 1.4.1 on your new device.

zigad, 2022-01-26:
I can confirm, on version 1.4.1 I can successfully receive all POCSAG messages (even restaurant pagers) since the latest update (with pocsag improvement) I cannot receive them. This is both for numeric and alphanumeric POCSAG messages with P polarity and 1200 baud-rate. 

 [image]
Picture is taken with 1.4.1 firmware with successfully decode of numeric / tone only pagers.

loewal, 2022-01-26:
Went back to 1.4.1
Tried all 8 combinations (p and B) at 172.450
On my Mac I recieve a lot, but a few weeks ago also on my Hackrf.
The red bar turns into blue, indicating that there is some reception.

loewal, 2022-01-26:
[image]
[image]

Nothing wrong with reception

zigad, 2022-01-26:
Good Evening Guys,

Today I went to nearest restaurant where they have those disc pagers (numeric/tone only).

Version 1.4.1 was able to decode every single one of POCSAG messages, but version 1.4.2 and 1.4.3 failed to decode every single one. I recorded few of POCSAG messages via capture app so I'm attaching it here: [POCSAG Messages.zip](https://github.com/eried/portapack-mayhem/files/7944443/Pocsag.Messages.zip)

Then I tuned to Slovenian frequency for alarming fire fighters via POCSAG protocol again 1.4.1 was able to decode crazy amount more messages than 1.4.2 or 1.4.3.

Settings were always the same during testing, antenna was the same, distance from transmitter was also always the same (around 15m for restaurant pager and few km away for fire department pagers)

ArjanOnwezen, 2022-01-26:
Hi @loewal & @zigad  thanks for reporting. It looks like #426 doesn't work well for some use cases and it might need a rollback. What do you think @eried ? I'll also try to contact @heurist1

loewal, 2022-04-07:
Hi all,
I am trying again...still no pocsag, using the latest firmware on H1: v1.5.0
On my iMac, using SDRangel, with the same antenna, I receive lots of messages.
Using 172.450, Dutch frequency.
Helas....nothing
Can anybody compare?

zigad, 2022-04-26:
Hi @loewal I can confirm that POCSAG RX still doesn't work as before. I looked and it seems that they didn't revert POCSAG app to previous version yet so no changes here. It's still on latest version that doesn't decode POCSAG messages as good as it did before 😕

loewal, 2022-04-27:
Hi Ziga,

Indeed, still not working (very well).
I noticed improvement, when I set some parameters in the Audio-section: NFM, full gain for all.
Then back to Pocsag.
Sometimes a whole lot messages appear, but also very often nothing at all.
@jLynx confirmed this and would report back. Until now he has been quiet for some time helas.

Let’s wait...

Regards,
Loe

ArjanOnwezen, 2022-04-29:
All, I've asked some people to test on discord in #pocsag channel. However others don't see to notice (big) differences. Could you also join that discussion

ArjanOnwezen, 2022-05-13:
Modulation has been hardcoded in in pocsag app now #644

loewal, 2022-06-09:
Did some additional tests:
- go to receive
- go to audio
- select (my Dutch) frequency of 172.450 MHz

Result: white noise in NFM, no Pocsag-signal at all

- select frequency of Dutch amateurs: 430.125 MHz
- only white noise

Started SDRangel and select channel Pocsag

Looks very nice and many messages....

[image]

closed 2023-04-26 by gullradriel