portapack-mayhem/mayhem-firmware #480: Add Gain_TX control to the GPS sim app

beged, 2022-01-26:
it was already reported in #395  and #446 and corrected the TX Gain control in the replay app,
however please correct it in the GPS simulator app too.
This is practically a file replay, so the same change is required there too.
In the GPS app still LNA gain can be controlled which has no effect to the out power, but here we need the TX Gain control too.

Brumi-2021, 2022-01-26:
Not familiar  with that GPS app , 
but I will try to have a look on  next week end. (As you said , maybe similar solution )

beged, 2022-01-27:
thank you, you will see it is same as replay, just list file with .c8 extension and look after a .txt file with the same name including the frequency and sample rate, otherwise a replay functionality
and the gain control should be the same as it was corrected in the replay recently

Brumi-2021, 2022-01-27:
Yes I applied the previous PR about Replay App , therefore I am familiar with that code part , and I will check it . But where can I get some GPS files to play with it , .C8 and its metadata .txt ? There are any samples I  the SD card folder or can you share me some file to make trial test?

beged, 2022-01-27:
yes in the standard sd card content you can find two files in GPS directory:

Directory of D:\_install kits\portopack-mayhem\files\GPS
04/14/2021  08:27 AM    <DIR>          .
04/14/2021  08:27 AM    <DIR>          ..
05/06/2020  11:00 PM       104,000,000 CATHEDRAL.C8
05/02/2020  03:17 PM                48 CATHEDRAL.TXT
05/06/2020  10:56 PM       104,000,000 EASTER.C8
05/02/2020  03:17 PM                48 EASTER.TXT

these are the shipping examples for GPS signal generation.

Brumi-2021, 2022-01-29:
Hello I already implemented  locally both previous PRs (same as Replay App)  , and I tested, and all works  ok.
The TX  RF level can be now  adjusted  by both settings  , GAIN and Amp .(exactly same as Replay App).
I confirmed  change levels with another SDR receiver  spectrum.👍
  
I also tried to see how it works,  with a close mobile with GPS ON , and replaying  those GPS_sim files . The mobile receiver GPS detects the simulation of several GPS satellites, (I can see simulated constellation)  but I think , just playing  20 secs are not enough to display different coordinates ,not ? I also tried replay loop for several minutes , but also not success. 

Later I will submit that  PR to be merged in next release.
By the way ,do you want to test it now  ?
If you want I can  upload now that test binary to discord ,in  test drive area .  Just let me know .

beged, 2022-01-29:
thank you @Brumi-2021 yes, the 20 sec records may not be enough... depends on phone too... but usually longer is required, and there are several other factors, like you need a TCXO ref GPSDO ref externally injected for very precise frequency and so on...
if you confirmed the level change with other receiver or analyser this seems good validation of the solution.
I am happy to test it, let me know the download link and I will do

Brumi-2021, 2022-01-29:
Usually , we use  discord ,   channel ,  "test-drive" to upload advanced testing binaries , to check correct functionality in all platforms ( H1,H2, audio AK4951, WM 8731....,  no side effects,...

The direct link is that one
https://discord.gg/tuwVMv3
But this time , I think , not so complex  change  , and it was a clone of the previous Replay App PR's ,
and I already confirmed , basic funtionality ,   then in my opinion ,  low risk to have side effects.
But in case that you still want the advanced version in discord,  let me know and I will upload it for you.
BR,

beged, 2022-01-30:
ok thank you, I agree that this is a low risk PR for side effects, and you made a validation test with detailed measurement,
I have only a H2 and no other hw, but please share the .bin in discord for my test
br

Brumi-2021, 2022-01-30:
Ok, to @beged , I already done , uploaded in the Discord,  (Test-drive area) : the advanved test binary , (based on current eried "next" + this PR)  , :   portapack-h1_h2-mayhem_GPS_gain.bin     (be awared , when you flash it , that  I added in the name . "_GPS_gain" ) .  Feel free to test it ,  and report , here.  Hopefully no other errrors.