portapack-mayhem/mayhem-firmware #492: read temperature sensors

theck32, 2022-02-20:
have a menu in the hackrf one portapack to be able to read the temperature sensors!

sivaelid, 2022-02-21:
The request is not clear. Do you mean the internal Temerature sensor thats located on the MAX 2837 chip .If so then this is monitored  and in options . You need to read https://github.com/eried/portapack-mayhem/wiki/Dissecting-the-Temperature-logger.
If you are talking about an external sensor then  you will need to fully specify to review  the characteristics and see if there are decoders/App avalable.

ArjanOnwezen, 2022-02-22:
Hi @theck32 It's already in the debug menu . However sensor isn't very accurate. See also link above

theck32, 2022-02-22:
Sorry for not being clear I speak French so my English is not very good, I am talking about the outside temperature sensors on the 433 mhz and 900 mhz to be able to read the temperature of its sensors which emits on its frequency bands (433.900 mhz) thx

ArjanOnwezen, 2022-02-22:
So you mean sensors of weather stations?

theck32, 2022-02-22:
Yes!

breakingcraft, 2022-03-01:
Something like https://github.com/merbanan/rtl_433 would be nice. Sadly I can´t any programming at all. Just an idéa.

eried, 2022-03-01:
duplicate of https://github.com/eried/portapack-mayhem/issues/113

closed 2022-03-01 by eried