portapack-mayhem/mayhem-firmware #494: App Title Description inconsistant to main menu -bug

sivaelid, 2022-02-23:
**Describe the issue**
There are different titles used on the main pages to that used on the heading of the App page. Some are confusing, or not fully understandable as they are in not in English,  where the rest of the applications are in English. The list of Bugs is below.

1. Main menu> Tools:   The heading is Utilities. Title could be Tools.
2. Main menu> Options: The heading is Settings. The title could be Options.
3. Main menu> Call: The heading is Search. The title in main page could be Search and ties up with Search list.
4. [Main menu > ]()Receive>Audio: The heading is Analogue Audio. The title could be Audio.
5. Main menu >Transmit> BHT xy/EP:  Is in French not English.
6. Main menu >Transmit> LGE Tool:  Is in French not English.
7. Main menu > Transmit> Burger Pgr: The heading is Si44XX TX. The title could be either.
8. Main menu > Transmit> TEDI/LCR: the heading is LCR TX The title could be LCR TX.

**To Reproduce**
Steps to reproduce the behaviour:

- Go to the main menu and select the items above to see the differences between the menu App title.

**Expected  behaviour**
That the descriptions in Main, Receive or Transmit menu, match the App tittle.
That the Language of the Apps are kept the same (The most common is English).

**Affected versions**
Latest release (1.4.3)

**Additional**
That the description suggested above need to be agreed in the community.

GullCode, 2022-02-23:
_"3.Main menu> Call: The heading is Search. The title in main page could be Search and ties up with Search list."_  
=> except that it's not a Search app.  it should be called 'Call'  even in the source code, and it should be Call list.
It has already bogged me as I was doing a real 'Search' app on top of the existing Scanner. I had to rename my classses, funcs and all because the Call app is not using it's terms correctly ('Search' everywhere inside instead of 'Call').
I even proposed a PR with a correct naming scheme for the Call app. Maybe you can find it in 'closed' or 'rejected'.
I have no opinion regarding the rest of your propositions ;-)

ArjanOnwezen, 2022-04-27:
Issue resolved in https://github.com/eried/portapack-mayhem/commit/ee9b4c89bdaf4b44a5ba50d27e7756e4912560b1 . Thanks @zigad.
@sivaelid If you also want to French to English translation, please open a new ticket. It's always good to split issues/requests.

closed 2022-04-27 by ArjanOnwezen

GullCode, 2022-04-27:
> Issue resolved in ee9b4c8

I can easily translate from French to English. Tell me :-)

ArjanOnwezen, 2022-04-27:
> _"3.Main menu> Call: The heading is Search. The title in main page could be Search and ties up with Search list."_ => except that it's not a Search app. it should be called 'Call' even in the source [...]

Sorry.. missed your comment.
If it helps I'll rename the app in menu and title to Calls.

GullCode, 2022-04-27:
Well, what would be good would be
-sed Search by Call inside the source
-rename the files
If you want I can make a PR, but maybe @Brumi will have a word to say about that

ArjanOnwezen, 2022-04-27:
@GullCode that would be good. 
@Brumi can also comment off course.

ArjanOnwezen, 2022-05-13:
@sivaelid  FYI also translated the French apps in #643

commit dccc68a4 (2022-04-27): Merge pull request #610 from zigad/fix-title-inconsistancy

Fix #494 - Change App Titles
commit ee9b4c89 (2022-04-27): Fix #494 - Change App Titles

I choose what I think are the best Titles based on existing titles/class names and so on. There were also inconsistencies between TX and Transmit and RX and receive. I renamed them to shorter version TX and RX also added it as suffix where possible to make it clearer in what mode you are in. If you have any other title suggestions or changes please use Add comment on Files Changed Screen so I can change it.