portapack-mayhem/mayhem-firmware #495: Microphone App

voxo22, 2022-02-23:
Mike-App is definitely major improvement thanks to Brumi-2021 in latest release 1.4.3. But, depth of modulation is still not very useful in AM/SSB, comparing to NFM. I can hardly hear myself on other hackrf with mayhem, despite I am using very good quality microphone, not just crapy headset mike. On the other hand, AM receiver (AUDIO mode) is perfect on mayhem. So the only problem is Mike app. I have tested AM/SSB modulation with mike also using SDRangel PC app, where I can fluently set any modulation depth and output signal is perfect and strong in combination with hackrf, like when I listen to any AM broadcast radio. Is there possibility to improve it somehow more?

There is also bug from 1.4.0, Mike-app does not transmit CTCSS in any mode and there is also no Roger beep. In 1.3.1 it does.
To make mike-app completely cool, there is need to add the possibility of AM/SSB receive option (half duplex thru TX button), not just only FM. Hackrf will then be possible to use like standard transceiver also for amateur bands (160/80/40/... meters)
Great work guys, keep up the good work!

Brumi-2021, 2022-02-23:
Hi thanks , now I am working on #465 as a additional feature for AK4951 users (ALC works very well , as a nice handsfree audio compressor , but Wind noise cancellation with  negligible effect , now struggling with equalizer) but I also saw that there is no roger beep , and I was wondering if never was there 🤔 ...  by the way which IC audio codec do you have ?  If AK4951, stay tuned , soon I will release advanced binary testing 
73

voxo22, 2022-02-24:
Yeap, I have AK4951. Thanks for your comment!
BTW, I do not miss Roger beep, but definitely Tone key. I have used hackrf to communicate with repeaters, without CTCSS, you know, is useless...
73

Brumi-2021, 2022-02-24:
Hi , I did not try before that  Tone Key CTCSS feature in TX.  But if it was working well before , as you said ,in 1.3.1 , it should be a regression possible to fix  . Do you know if all those two features (roger beep and CTCSS) were also available in WM8731 codec version ?  Anyway after finalising with ALC issue, I will have a look . 
73

Brumi-2021, 2022-03-01:
Hi @voxo22 , I confirmed , as you said , that in the Mic App , the “roger beep” and the “CTCSS= Tone key”  were  working well in the previous version 1.3.1 , and we lost them from version 1.4.0 onwards. (When it was introduced the other excelent different modulation TX feature of the AM/DSB/USB/LSB/FM ).

Testing 1.3.1., I could see the spectrum of both features and also hear  them in another receiver.  I understood how they should behave. 

In current Mayhem v 1.4.3 version , (same as 1.4.0, 1.4.1, 1.4.2) : 
*  the  option of the “roger beep” , can be selected through menu in the GUI , but it does not work .
* The CTCSS option can not be selected  by menu (or I could not) , so no way to test it . 

Now I almost finalising #465 , activating Digital Auto Level mic control (in AK4951 audio codec version) in the same Mic App , and  definitely , after that I will have a look on it , trying to recover both CTCSS Pilot tone and the roger beep. 

73

Brumi-2021, 2022-03-01:
Hi again  @voxo22 ,  so far , now in my local test code, menu Tone Key selection works again   (but TX spectrum ,  still does not works) .  Step by step.

Brumi-2021, 2022-03-09:
Hi @voxo22 ,
 I already finished #465 (if you can , pls also test it , beta binary is in test-drive discord ) .in next days ,  I will continue investigating why we lost CTCSS feature , and trying to recover it !
73

Brumi-2021, 2022-03-20:
Hi again @voxo22 and all ! 
After bisecting many commits changes and navigating back through the code , I already found where is the problem where we lost those ham  features ! 
The code to make key tones CTCSS and the "roger beep" is there also inside 1.4 3,  and it works well ! 
 But it does not work with all the latest changes all together . To my eyes , it is quite complex Mic App , but really nice one !!! and quite well structured and logical. 
 So far I still could not solve it, but getting closer ! 
Hopefully it will be fixed soon 🤞

voxo22, 2022-03-20:
Thank you! While reviewing the code, will you also add AM/SSB receiver mode in Mic App as I mentioned earlier? :)

Brumi-2021, 2022-03-20:
Yes I also would like to have  it !!! but Sorry I think that this would be too complex to me (at the moment )...
 I think we will need extra help from the top skilled masters ...

  Adding AM (to the FM) receiver it migh be easy , But for adding  SSB receiver , maybe we will need to consider , how to manage GUI interface ?  adding spectrum waterfall in the LSB , USB .???.,   If we do not need FFT waterfall , maybe achievable ,...  but  we will need extra help from the original Receiver App developpers . 
  73

voxo22, 2022-03-21:
Ok, that's fine, I understand :-) Personally, I wouldn't reuqire to do any major changes in GUI, once you know the frequency, it's enough to set it directly. I mean just the same way as in Tx section... FM/DSB/LSB/USB... So we will wait for the Gurus to possibly adding something to this app :)
73

Brumi-2021, 2022-03-21:
Ok , if no major GUI changes request  , nor FFT spectrum waterfall , maybe possible to handle  !!! (Because the code is already there , we just need to re-use it in that App)... (but of course always  better with the extra help of the fw Portapack Gurus !!! ) . I still di not have the full firmware architecture picture (but learning day by day ).
Anyway , let's go step by step .
My next challenge would be recover the lost features of key tones (CTCSS) and roger beep in next PR ! 

By the way did you also  try  my previous binary about Automatic MIC volume level control (ALC) ? (Binary is in the test drive discord channel)
73

Brumi-2021, 2022-03-26:
Hi @voxo22 ,and all, 
I already managed to recover the "Key toness-CTCSS" in the 1.4.3. To my eyes, that change will be LOW RISK .
I am activating it only in FM (no meaning AM/SSB)

(Still pending to solve the "roger beep"), that is on going, second Priority.

How to proceed ? 
Shall I just apply independent PR about it or shall I modify my previous one, related to ALC AK4951 #465  and combine both  ?  (That was  bigger change , but I also think low-mid risk of side effects)
In any case we will need and appreciate your testing and validation.
.
Shall we combine and release a beta testing both things integrated or just  another 1.4 3 with this  CTCSS? 
73

ArjanOnwezen, 2022-03-26:
> Hi @voxo22 ,and all, I already managed to recover the "Key toness-CTCSS" in the 1.4.3. To my eyes, that change will be LOW RISK . I am activating it only in FM (no meaning AM/SSB)
>
> (Still pending to [...]

@Brumi-2021 usually smaller changes are better, since they are easier to code read. 
However if you already changed many thing in the same program, it might be easier to include CTSS as well.

Brumi-2021, 2022-03-26:
@ArjanOnwezen yes , I fully agree with you . 

So far this CTCSS  bug fix  on 1.4.3  is smaller code change ,and low risk to have side effects . 
I will try to combine Key tone -CTCSS bug fix + roger beep (if I can also solve it . It will be my next challenge) ,
Then we will recover both lost features from 1.4.0 onwards. 
If I I really struggle with roger beep ,then  I will release separated PR releases (Key tones subcarrier one ,  and Roger beep another one )
My plan 
Key tones --> only activated  for FM .
Roger beep --> for all modulation modes ,FM /AM/SSB .

 (And lets leave ALC AK4951 PR  evaluation separately... they are totally independent features , even same Mic App)

ArjanOnwezen, 2022-03-26:
Cool! Keep up the great work @Brumi-2021

Brumi-2021, 2022-03-31:
Hi , just to share the status .₩

voxo22, 2022-03-31:
@Brumi-2021 , I have tested new 1.4.4 release with ALC. I have tested FM/AM/SSB/DSB. ALC seems to work fine, but unfortunately, I still have problems with Tx sound, modulation is very weak (no matter what level meter on screen shows), and what I have forgot to mention in my first note, there is quite strong whistling background noise in carrier in both FM and AM Tx mode.  Is that normal? I mean normal for portapack? I did not feel it so annoying using hackrf in usb mode (sure it is there, but not so intensive).
Testing suite: 1 portapack for Tx, 1 portapack for Rx (both H1/AK4951).
To make things clear: what kind of microphone do you use for the testing? Is it one on headset, or standalone? Condenser or dynamic?

Brumi-2021, 2022-03-31:
Hello @voxo22 , thanks for your testing . 
How do you check modulation level? I was using Tecsun receiver WFM , but in that case you need to increase the FM deviation modulation (called wrongly BW)  from 10 --> 75 . and to me WFM and FM Looks like ok very good index modulation  . I also have more professional ham radio receivers , but I did not test it there .
Regarding AM and SSB I intentionally limited the audio output of the ALC  out to avoid harmonic TX  distortions, because in Mic App all those modulation are treated in Hilbert transform filter , and maybe (just my guess) it is limiting us ..  (normally in AM case in GNUradio it is not using Hilbert, Hilbert it is usually used in SSB)  but here , in our fw , it is also used in  a clever way ! . 
You can see easily with spectrum RTL SDR receiver those harmonics when overmodulation. Regarding my MIC i used an old iphone 5S accessory  (i annexed the photo, i believed electret type , good quality ) 
[image]

By the way regarding CTCSS that was working in 1.3 1 with FM only  , I did not give up ! Still investigating. 
I managed to have a correct working local branch from commit equivalent to tag 1.4 0 with FM/AM/SSB but when I applied those modifications to the current 1.4 4 I got strange frequency tones .... I am missing some other modification PR or constant factor ... that breaks my code ... I believe that I am  very close to solve it in 1.4 4 . (But I am ignoring  the roger beep by now , second priority) . From 1 4.0 to 1 4.4 there are hundred of code modifications... then probably I would go step by step , applying my sw code changes in 1.4 1 , 1.4 2 ... and try to find were is the conflict . 
73

Brumi-2021, 2022-04-03:
Hi @voxo22 and all,  
Pls  have a look into the latest  above PR  and  go to the discord Test-drive and download the beta_CTCSS binary .

73 ,

Brumi-2021, 2022-04-06:
Hi @voxo22 , have you had time to test in a real ham radio repeater the Key tones -CTCSS ? 

In next days I will try to also to re-activate the lost feature  roger beep (and the real  GUI MIC GAIN feature ,now only modify the Mic vu-meter but not the real modulation level).  But before that , it would be appreciated your key tone -CTCSS validation test. Thanks in advance 
73

voxo22, 2022-04-07:
Hi @Brumi-2021, I had some free time two days ago to thoroughly test the Mic app. Unfortunately I haven't had a chance to test the CTCSS feature yet, but I'll give it a try as soon as possible. Brumi, I still have a problem with the microphone itself though, I'm testing the app with an external standalone microphone, with a Jabra branded headset, also with a similar cheap headset to the phone that you took a picture of. I feel that your ALC, which you coded recently, has excellent gain, but the actual peak cancelling effect from the microphone is only minimal. It fails to prevent premodulation (annoying scratchy distortion) of the signal, even though the level bar is well below the red zone. The distortion occurs even with the ALC turned off. Throughout the application, I miss the consistency between the GAIN setting, the ALC setting and of course the link to the level bar, as the latter should never reach the red when the ALC is activated, and thus not allow distortion (overmodulation) of the output signal.
Personally, I would leave your ALC present in the app, but I would throw out the GAIN setting altogether, it doesn't make sense there at all. ALC also substitutes for GAIN, but the compressor should be more efficient, it must not allow to reach a level that causes distortion at the output. An application modified in this way would be perfect. Later today I will try to upload a video describing my test.

Brumi-2021, 2022-04-07:
Hi @voxo22 , thanks for your feedback.
I am also not using current Mic Gain . I am using ALC.(only available as additional feature to the AK4951 users)
But we need to keep and improve current  common adj Mic Gain  For the rest of users that has WM8731. 
Current fw is only applying Mic Gain to the vu-meter , not to the original voice modulation (as it was in fw 1.3 1) it needs to be fixed .
In next days I will try to fix this incorrect Mic Gain and the lost Roger Beep . 
Pls feedback to us about CTCSS key tones recovered feature . We will need to find the best compromise with @teixeluis  for his Sondes App . (Because to reactivate correctly the CTCSS I need to bypass some small part of his previous changes )
Let's keep in contact !
73

Brumi-2021, 2022-04-16:
Hi @voxo22 , and all,  
 I have some good news to you (but STILL  BETA TEST VERSION under development) ,

I am still working on it .  But quite good progress , 

Hi,  let me upload a beta under dev.  advance beta test Mic App ,  recovering FM-CTCSS ,  Roger beep , and correct x Mic gain scaling .  Still pending to try to activate it on SSB (LSB / USB) ,  step by step .  It is based on 1.4.3 (including ALC on AK4951 platform) .  I think  that now working well  FM / AM / DSB  ,    (pending SSB) .
This is my summary status photo
[image]
Do not forget , that I added in the name , _beta_CTCSS_BEEP_pend_SSB.bin   

It is under development , but pls. test  ==> FM / AM / DSB part  and  let me know about CTCSS / ROGER BEEP / Mic Gain 
  (SSB still  pending,  and   it may crash if you activate BEEP and  still no vu-meter adjustment yet )

I upload the beta binary (portapack-h1_h2-mayhem_beta_CTCSS_BEEP_pend_SSB.bin)
 ,  in discord HackRF Portapack Mayhem  channel : test-drive.,   (but remember , IT IS JUST  BETA UNDER DEV not finalised) 

73

GullCode, 2022-04-16:
Good work, I'll test if I have some time tonigh

Brumi-2021, 2022-04-16:
> Good work, I'll test if I have some time tonigh

Thanks @GullCode ! 
Later on once finalised, I will  also  explain about it ,  asking to sw masters and gurus from the Community  to help me to review the code and optimize it .

Let me know about your test for FM / AM / DSB , 
Playing with Mic gain factor in all those Mod types
 Roger beep in  "  "  "  " 
and  CTCSS only  in FM .  

Thanks

Brumi-2021, 2022-04-27:
Hello @voxo22 and all,  

pls find attached beta testing , re-activated Key-tones-CTCSS / Roger beep / and correct Mic GAIN application . Key-tones-CTCSS is only available  in FM (because no meaning in AM/DSB/SSB) ,  Regarding Roger beep  is available in FM / AM / DSB , (I could not manage in SSB) .  I just  checked in my platform   H1 -AK4951  platform and seems all OK (I know that in some H2 WM we have still some no sound  problems , pending to be addressed) .

  I need to change back some part of previous @teixeluis PR's  about his latest changes in  tone gen used in Radiosondes. (In this Mic App , we are using same function, but  it is very critical the CPU code optimization, and the original code function, was more optimized  ) . We need to test and check carefully the sound code in the Radiosondes app when getting higher RSSI signal after that change back, ,( pending to be checked the radiosondes sound effect) .  

Let me add a more visual  summary of the latest  changes ,(I hope no errors ) 

[image]

 If no  other known bugs are  reported,  soon ,  I will release the PR .
You can download the beta binary :  portapack-h1_h2-mayhem_CTCSS_Rogerbeep.bin in the  Mayhem discord, channel Mic-App.      waiting for your feedbacks . 

73,

Brumi-2021, 2022-04-28:
> half duplex

By the way @voxo22 . 
I still did not review that other reception part  code, but accidentally I could hear headphone sound demodulateion difference when changing on top menu the TX mode FM/AM/USB/LSB/DSB .  Then I did not test by my own , but maybe what you were requesting (RX in FM/AM/SSB)  was already done?  Not sure , but 
Did you check it , (I still did not ) 
73

ArjanOnwezen, 2022-05-07:
This issue is solved by @Brumi-2021 in #633

closed 2022-05-07 by ArjanOnwezen

voxo22, 2022-05-27:
@Brumi-2021  I have recently installed release 1.5.3. What happened to ALC in Mic app? Where is it?

GullCode, 2022-05-27:
I think it's not yet merged

Brumi-2021, 2022-05-28:
Hi @voxo22 , (and all).
YES after confirming that in some H2  WM8731 platforms  it produces no Mic App sound  , we temporary removed , when I added back CTCSS , and roger beep .
Now that I have additional H2+ WM8731 QFP100 sample , I will debug , try to solve full compatibility with WM8731 and will apply a new PR merge . It should be transparent to WM8731, with no side effects. Sorry for related inconveniences 
Stay tuned 
73

Brumi-2021, 2022-05-28:
Hi @voxo22  (and all) 
I just  created the related PR, 

I think it is solved , I checked in both platforms and  seems OK,  (the root cause it was clear ,and easy to be fixed ) 
But we will happy to  get  both  platforms  evaluations results ,   AK4951 (about ALC) /  WM8731 (about no more audio lost after using Mic App or inside Mic App ) .
Soon it will be merged, if no more side effects are found .

Thanks, 
[portapack-h1_h2-mayhem_153_ALC.zip](https://github.com/eried/portapack-mayhem/files/8792295/portapack-h1_h2-mayhem_153_ALC.zip)

Brumi-2021, 2022-06-01:
Hi @voxo22 , have you had time to check the above version ? 
If yes , pls add your comments below the PR https://github.com/eried/portapack-mayhem/pull/664
Thanks

Brumi-2021, 2022-06-13:
Hi @voxo22 , any news about it ?  (Or other testers in both platforms AK4951/ WM8731 are also welcome)

If yes , pls add your comments below the PR https://github.com/eried/portapack-mayhem/pull/664
Thanks