portapack-mayhem/mayhem-firmware #496: Support for Portapack H2+ with new QFP100 CPLD EPM240T100C5N

goatyde, 2022-02-25:
Chinese portapack H2+ of latest version will just briefly flash and then not display anything when updated to latest mayhem version.
Example model: https://www.ebay.de/itm/284658778941

Support new CPLD used as described here:
https://www.mmdvm.club/index.php/archives/197/
(Seems like a hack to me, not the proper solution, but what do I know)

eried, 2022-02-25:
oh no, another hw version! 😅

goatyde, 2022-02-25:
I´m so sorry I bought it ;-)
I could try to start programming that myself, but my time is limited as I´m doing full-time programming already and electronics in the evening. I tried to put a request/comment on the chinese page to encurage some input from them.

zxkmm, 2022-02-25:
> oh no, another hw version! 😅

It's not another. It is just that one which we talked a lot in ur Discord server. Which jumbo77 had modified an "at least bootable" firmware although contain a lot of bug. And what jumbo77 did is just skipped the CPLD check step while booting.

zxkmm, 2022-02-25:
> I´m so sorry I bought it ;-) I could try to start programming that myself, but my time is limited as I´m doing full-time programming already and electronics in the evening. I tried to put a request/comment on the chinese page to encurage some input from them.
> 
> It´s possibly already handled in this: #421

Hello dear goatyde, thank you for your work! I'd like to coorpration to fix this with you if ud like me to. I got a half year vacation so i have bunch of free time. And you my mail address is $my github user name$@hotmail.com. Thank you!

goatyde, 2022-02-25:
What work ? I am simple user, and I have no clue about the PortaPack so far. 
If you have a clue about how to implement the correct detection of the CPLD or what is needed, then you probably should submit a patch or something here. Or make a pull request like the #421

zxkmm, 2022-02-25:
> What work ? I am simple user, and I have no clue about the PortaPack so far.
> If you have a clue about how to implement the correct detection of the CPLD or what is needed, then you probably should submit a patch or something here. Or make a pull request like the #421

Ah, mate I saw you said you’ll program urself to solve that cpld problem. Me and a guy in Discord are trying to fix this already but no progress for now, so what I mean is we may make a cooperation ( pure non-profit cooperation ) and find bugs in [this](https://github.com/jumbo77/portapackh2/issues/3#issuecomment-1013206285)  firmware and fix it. Btw when I say “ty 4 ur work” I mean future work :)

sivaelid, 2022-02-25:
Hi I have a PortaPAck H2+ but it came with QFP64 and works ok  . The https://www.mmdvm.club/index.php/archives/200/  has a modified firmware 1.4.2 for you to down load  and use . Not sure what they have done in the mod??

zxkmm, 2022-02-26:
> Hi I have a PortaPAck H2+ but it came with QFP64 and works ok . The https://www.mmdvm.club/index.php/archives/200/ has a modified firmware 1.4.2 for you to down load and use . Not sure what they have done in the mod??

The firmware you are using is exactly jumbo77’s version. And what jumbo77 did is just skipped the CPLD check step while Mayhem is booting. This method is more like a temporary method. They just changed 2 line of code. And didn’t fix other potential problems. You may read the code [here](https://github.com/jumbo77/portapackh2/issues/3#issuecomment-1013206285).

Aheister-Prox, 2022-03-11:
I also bought one of these Chinese-made PortaPack H2+ and at first, I was stumped/struggled on why I couldn't get it to work, then found the [JUMBO77Ver1.4.2]( https://github.com/jumbo77/portapackh2) version which got me started at least. I've had a few issues with the 1.4.2 I've noticed right away. The ADS-B mapping doesn't work (something about missing map.bin file) and on the PortaPack hardware side during audio receive every time I press one of the buttons (up, down left right) you can see this in the waterwall as a signal distort. Anyways it would be nice to have 1.4.3 running on it without code mod so if someone can make that happen it would be greatly appreciated on this end.

zxkmm, 2022-03-11:
> I also bought one of these Chinese-made PortaPack H2+ and at first, I was stumped/struggled on why I couldn't get it to work, then found the [JUMBO77Ver1.4.2](https://github.com/jumbo77/portapackh2) [...]

If refund possible, make a refund. But anyway the source code of 1.4.3 is here. U may build it. https://github.com/jumbo77/portapackh2/issues/3#issuecomment-1013206285

zxkmm, 2022-03-11:
> I also bought one of these Chinese-made PortaPack H2+ and at first, I was stumped/struggled on why I couldn't get it to work, then found the [JUMBO77Ver1.4.2](https://github.com/jumbo77/portapackh2) [...]

It seems not quite possible. If refund is possible, pls make a refund. But signal distort is unexpected even with new cpld. Did you install the emc shield on your hackrf?

felipecaon, 2022-03-27:
Just got a h2 portapack with EPM240T100C5N, flashed and installed 1.4.3 jumbo version. In his version the update if statement is commented as @zxkmm said. Wouldn't be the case were we could check the CPLD config before jumping into the if statement? 

I'm quite new in this kind of stuff but wouldn't be hard to make things co-exist, I'm going to play with h2 firmware during the week and see if I can figure out a way to implement this decently

QioTek, 2022-04-04:
I'd like to know Chinese-made clone portapack H2 + with new qfp100 CPLD epm240t100c5n board can be supported in the firmware? If can't support it, I'll refund it. I need a clear answer, thanks.

ArjanOnwezen, 2022-04-04:
If it doesn't work with current firmware, ask for a refund. Impossible to support all boards.

QioTek, 2022-04-05:
At present, I can upgrade to the 1.4.3version(https://github.com/jumbo77/portapackh2). After upgrading, I don't find any issues, but I want to know whether I can continue to upgrade in the future. @ArjanOnwezen

ArjanOnwezen, 2022-04-05:
Nu clue @QioTek. Please check with author of https://github.com/jumbo77/portapackh2.

ArjanOnwezen, 2022-04-06:
Could everyone facing issues try https://github.com/eried/portapack-mayhem/releases/download/nightly-tag-2022-04-06/mayhem_nightly_n_220406_FIRMWARE.zip

Closing the issue.

closed 2022-04-06 by ArjanOnwezen