portapack-mayhem/mayhem-firmware #498: Recording audio: filename = frequency + datetime

voxo22, 2022-03-09:
Browsing AM bands from 0 to 30 MHz I often hit some interesting sounds which I want to note. Recording audio is very useful for this. But recording misses any info, its very hard to remember later when and where it happen. Instead of numbering the files, would it be better to have filenames with frequency and datetime info, or, at least some kind of additional file with them (as in capture app).

eried, 2022-03-10:
Very good idea

notpike, 2022-03-29:
Hi,

Just made a pull request for the enhancement above. This dose make the file name a tad too large to read within the UI but it doesn't break the file manager or wave viewer. The format I chose was ISO 8601 and the file names will be saved as "AUD_YYYYMMDDTHHMMSS_0000000000Hz.WAV" .

Let me know if you think this viable or not. :3

ArjanOnwezen, 2022-03-30:
Implemented by @notpike

closed 2022-03-30 by ArjanOnwezen

voxo22, 2022-03-31:
Hi @notpike , thank you for this! Just only one problem is with showing the filename on display (AUDIO Rx), while recording, it messes up the line with % and free recording time (characters are overlayed with these parameters). Maybe you can hide the filename, or display it anywhere else on the screen (bottom of the waterfall?)