portapack-mayhem/mayhem-firmware #499: POCSAG-TX does not transmit data stream

Titus-770, 2022-03-09:
Bug : 
When using MAYHEM POCSAG transmit software the HACKRF keys up but does not transmit any POCSAG data. 
This does not happen when using the same gear in HACKRF mode via GNU-RADIO & GR-MIXALOT

Equipment : 
HackRF, Portapack H1
MAYHEM v 1.4.3
GR-MIXALOT 
DragonOS_Focal 
Baofang UV-5R
Motorola Advisor classic pager 

Steps to reproduce the fail behavior:

1. Start up MAYHEM, tap TRANSMIT then POCSAG
2. Settings : 
    - Bitrate : 1200 ( Also tried 512 ) 
    - Address : 1111112
    - Type : Alphanumeric
    - Function : A
    - Phase : P
    - Message : Portapack
    - Frequency : 152.2400
    - Channel Spacing : 25k
    - Shift : 9kHz ( +/- 4.5 kHz )
    - Gain : 47
    - Amp : 0
    
3. Press start
4. Monitoring the frequency on my Baofang , the squelch opens and the transmission begins , but no POCSAG data is sent. It's just silence until the transmission stops. 
5. No response from pager

Steps to reproduce successful behavior : 
1. Power up HACKRF with MAYHEM firmware
2. Select HACKRF mode, display changes to blue status screen
3. Boot dragonOS
4. Load GNU-RADIO
5. Load script provided by GR-MIXALOT : pocsagtx_hackrf.grc
6. Configure as above 
7. Click on RUN
8. HackRF display indicates transmission in progress
9. Baofang radio squelch opens, and you can hear the data being transmitted
10. Pager lights up with notification beep and displays message
11. In GNU-RADIO select GENERATE
12. From shell execute generated python script
13. HackRF display indicates transmission in progress
14. Baofang radio squelch opens, and you can hear the data being transmitted
15. Pager lights up with notification beep and displays message

**Expected behavior**
Using the MAYHEM TANSMITTERS POCSAG software, expected the POCSAG message to be transmitted to the pager, which will beep and show the message.

**Affected versions**
MAYHEM v 1.4.3
DragonOS_FOCAL
GNU-RADIO 3.8.1.0

**Additional**
I do have the required hardware and software to preprogram the pagers. I have had the same success and fail results at both 512 and 1200 baud rate. 
I have tested this with three different pagers with the same success/fail results.

smannchen, 2022-03-10:
I was using it earlier, and had no issues. Monitored using sdr angel, and an rsp1a. Verified data was sent. Also received fine with my Apollo pagers.

Titus-770, 2022-03-10:
> 

Thanks for the info. I was advised against the H2 due to supply issues like shorted out buttons and leaky batteries. Where did you get yours?

smannchen, 2022-03-10:
Got mine from TZT Trading on aliexpress. To be fair Chinese clones of anything often have quality control issues. This is my 2nd H2 from them though, and I haven't had any issues other that a broken button and that was my fault.

Titus-770, 2022-03-10:
> Got mine from TZT Trading on aliexpress. To be fair Chinese clones of anything often have quality control issues. This is my 2nd H2 from them though, and I haven't had any issues other that a broken button and that was my fault.

I'll see about getting one. I just think in my case it's odd that even with the h1 that it works when connected to a computer, but not when using the MAYHEM firmware by itself.

smannchen, 2022-03-10:
It definitely could just be an issue with the portapack itself if the hackrf is working fine.

Titus-770, 2022-03-10:
> I was using it earlier, and had no issues. Monitored using sdr angel, and an rsp1a. Verified data was sent. Also received fine with my Apollo pagers. Only difference is I'm using the H2.

Can you share the configuration of your MAYHEM POCSAG software, along with any other settings you use?

Titus-770, 2022-03-11:
SOLUTION : update CPLD as per : https://hackrf.readthedocs.io/en/latest/updating_firmware.html#updating-the-cpld

As per Mossmann : 

I believe that r7 was our first production since we stopped flashing the CPLD at the factory. The CPLD is now programmed by the firmware every time it boots. It is possible that Mayhem assumes that the CPLD has been flashed.
You can find the default.xsvf file in our git repo or our latest release package.
Our firmware ignores the CPLD’s flash configuration, so doing this won’t affect it, but I suspect it may affect Mayhem’s behavior.

closed 2022-03-11 by Titus-770