portapack-mayhem/mayhem-firmware #500: Problem upgrading the firmware

Moroliner, 2022-03-11:
Just bought a brand new hackrf(clone) and a portapack2 with Mayhem 1.3 preinstalled and once i tried upgrading the firmware using the steps written on the Guide it isnt working any more.
The Upgrade finished without any errors but when i try to start it the lcd quickly flashes white and then is just black and does nothing.
Any help would be gladly appreciated :)

Aheister-Prox, 2022-03-11:
I bought one of those Aliexpress Portapack H2+ knock offs and was had the same issue. I used [JUMBO77Ver1.4.2](https://github.com/jumbo77/portapackh2) which resolved that display flicking first issue. From there,  I was able to update to 1.4.3 [JUMBO77Ver1.4.3](https://github.com/jumbo77/portapackh2/releases/tag/FW3).

Follow the instructions exactly; (See links above as the steps were pulled from there)

1.HackRF Press DFU then Press RESET (then install DFU Driver)
2.Click dfu_hackrf_one.bat (then install hackrf Driver)
3.Then Click flash_portapack_mayhem_skip_cpld1.4.2.bat
4.update Finished

Hope this helps you.

Moroliner, 2022-03-11:
You are a Legend works perfect now
Thank you very much :)

closed 2022-03-11 by Moroliner

Jugrnot, 2022-03-15:
> I bought one of those Aliexpress Portapack H2+ knock offs and was had the same issue. I used [JUMBO77Ver1.4.2](https://github.com/jumbo77/portapackh2) which resolved that display flicking first [...]

I too have one of the ebay/aliexpress clones with 1.4.0 installed and has some glitches I'm hoping that 1.4.2 can fix..... the problem is, the PC recognized it in "hackrf" mode with software, but absolutely nothing I can do gets me to a point where I can update the firmware.  I've tried 5 different computers, some that this thing has never once been plugged into, and at least 20 different USB cables.

Checking in device manager, HackRF One comes up when you're in "hackrf mode" but holding the reset+dfu buttons, in any combination, for any period of time, the PCs never find the device -- or ANY new device for that matter.  Have invested at least 2 hours of my time on google/github looking for some solution, and I've come up with absolutely nothing.  Any suggestions??

Moroliner, 2022-03-15:
For me it worked like that:

-1st i removed the Portapack from the HackRF 
-2nd then I plugged the HackRF into my pc and then put it in the DFU mode cause when the portapack is still on the DFU doesn't work
-3rd and then I did as Aheister-Prox told me above and it all worked

Jugrnot, 2022-03-15:
> For me it worked like that:
> 
> -1st i removed the Portapack from the HackRF -2nd then I plugged the HackRF into my pc and then put it in the DFU mode cause when the portapack is still on the DFU doesn't work -3rd and then I did as Aheister-Prox told me above and it all worked

I was really hoping to not have to take this thing apart.. it's not remotely fun to do.  Anyway, took it apart, plugged in just the hackrf, DFU mode, ran dfu flasher and received this:

[image]

I can't even determine if this was successful or not?  Unplugged USB, put the portapack back on, plugged into PC and portapack still shows 1.4.0, and I'm still unable to run the portapack flash as it doesn't detect anything.  Go into device manager, update driver, pick manually form list, have disk, select hackrf_one.inf, okay, okay.  Try to run flash_portapack_mayhem.bat and continue to receive HackRF not found.

I'm skeptical it even updated the god damn driver:
[image]