portapack-mayhem/mayhem-firmware #501: DVR-Style Capture/Replay

HackingIntoYourHeart, 2022-03-11:
**Describe the solution you'd like**
I would like the ability to do something like:
While recording in the Capture app, “save the last 5 seconds of the waterfall” or last 10 seconds or 15 or whatever. Maybe there’s a setting that decides what the X is for a “save last X seconds” button 🤷‍♂️
OR (more practical):
The ability to clip recordings from the Capture app. OR the ability to Fast Forward or Rewind while replaying signals in the Replay app.
Essentially, I’d like to not have to see a signal appear on the waterfall and boom, now it’s too late to record. To record signals I have to preemptively know when it will be broadcasted. It’s frustrating and impractical. My solutions may not even be the best in your eyes, and that’s fine. It’s very possible that you have a better idea and that’s extremely alright! Either one or ALL of these ideas would be fantastic 👍

**Describe alternatives you've considered**
I have tried saving the files to the SD card, moving them into audacity, clipping, then moving them back, however no matter the format it breaks the new file. Plus, this is clearly impractical. Everything being native to the Mayhem firmware on the portapack itself is practical and preferred.

Let me know if i filled this out wrong or didn’t pay right? I don’t entirely understand how this request thing works. My bad.

eried, 2022-03-12:
I think the description is ok, basically you can assign a bounty here: https://app.bountysource.com/issues/105950068-dvr-style-capture-replay and then discuss with interested people in the comments here. If the feature gets implemented, then the money is released when the issue is closed

GullCode, 2022-03-15:
>  "Maybe there’s a setting that decides what the X is for a “save last X seconds” button man_shrugging"

Regarding the GUI and the easiest way to give that kind of functionality, it would be adding two settings in list buttons at the right of the recording button. I see it like first list button would be a range of seconds values, for example, and the second list button would be a range of buffer value for SD card writings (because depending on the hardware H1/H2/H2+ and SD card brand, some may experience better performances / less fragmentation if they can specify the block size / alignement).

Regarding clipping, well. Since some app are displaying a waterfall, one may be able to resize a capture so the waterfall of the signal fit the screen vertically, and allow a selection with two button (lower bound/upper bound) using the rotary encoder.
A scrolling along a zoomed version of a capture is doable but will need to do buffered reads / seek in file to keep memory usage low.

I hope some will jump in the mod, it's a good start for portapack app development learning :-)

Brumi-2021, 2022-03-16:
Hi , I do not want to be pessimistic... but recording BW 500 Khz , we are generating 2Mbyes / second in .C16 format  and unfortunately our ARM LPC4320 only has 200Kbytes RAM  ... therefore we can handle as a maximum  very small buffer (100mseg) at this speed ... then as a compromise maybe what could be done is trigger auto start recording of 10 secs (example) from the first RSSI > threshold (+ small miliseg buffer).   Normally any keyfob is sending the same command several times and you will get almost the first + all the rest.   
Those are my thoughts...

GullCode, 2022-03-16:
Well, a write buffert is usually a power of two, like 128,256,512,1024,2048,4096,8192,16384,32768,65536 and so on. Based on  my experience on SD card, it will be at most a few Ks before having no gain.
To save some CPU maybe we can reduce the quality of the waterfall if in 'last X secs recording' mode ?
If I have the time, I may make some tests.
@Brumi-2021 the idea was to use an indexed file on the sdcard. You say you want last X seconds, then allocate a X second sd buffer. We have to save starting and ending point because the file will be written like a rotating array. (I hope you get what I mean). Then selection/cut would just be adjustments of the indexes. To write signal, dump until reaching eof, move indexes and continue if needed. To save the signal, go from starting index, if you reach EOF start back at the beginning, read until you reach the end index. Copy byte by byte, or by buffered copy of a few Ks to save CPU/IOs.
It imply that it should be an option activated on demand, because it would permanently dump the signal as you watch the thing. It has to be tested, just to see if there is enough CPU to manage a SD card dump in the background while displaying a waterfall.

gullradriel, 2023-05-10:
After a good thinking, it's too much for the hardware.
Well, prove me wrong if not ;-)

closed 2023-05-10 by gullradriel