portapack-mayhem/mayhem-firmware #502: Brightness Control

TriniHam, 2022-03-15:
There should be a option to adjust the brightness. This would help with extended battery life. And wouldn't be as harsh on the eyes.

ArjanOnwezen, 2022-03-18:
Are background LEDs hardwired or not. If they hard wired, it is not possible to dim the background light without hardware hack.

TriniHam, 2022-03-19:
> Are background LEDs hardwired or not. If so, it's not possible to dim the background light without hardware hack.

Wym and I believe it's wired but LCD, and if it is wired can you give some light as to how you would go about this. I would like to attempt this myself but I believe for this to work we would have to call the screen brightness to some level ,  @eried is there any special keywords we must know to do this. I'm thinking when the Portapack times out there's a function that being called to do that. This is a example of how I interpret the Portapack to time out.

timeout(){
  if display < 3000;
     display 0;
  else display 100;
}

If my interpretation is correct then. To make it dim instead of saying display 0 we can say display 50 or whatever variable is comfortable .After that part is completed then we can work on the actual permeant feature. So @eried am I on the correct track and is there any special key words ?

eried, 2022-03-19:
You should run a test, just replace the 0 by 50 🤠 (I have no idea if there is some pwm, dac or just on/off backlight)

TriniHam, 2022-03-19:
> You should run a test, just replace the 0 by 50 🤠 (I have no idea if there is some pwm, dac or just on/off backlight)

😂 Ok so to look at the code and change values do I follow this link[this](https://github.com/furrtek/portapack-havoc/wiki/Building-from-source) or is there another way to decompile or a much simple way to extract the code and @eried do you know the location of where that file may be?

eried, 2022-03-20:
You can use docker, which is the easiest https://github.com/eried/portapack-mayhem/wiki/Compile-firmware

TriniHam, 2022-03-20:
> You can use docker, which is the easiest https://github.com/eried/portapack-mayhem/wiki/Compile-firmware

Sigh been having a hard time doing that on my system is there any good video demonstrating how its done? (decompiling and compiling). And the version on your github that's essentially the decompiled version right.

eried, 2022-03-21:
I have not seen a video of the process, I guess everyone stresses before it works and then forgets after it does so nobody has taken the time to document it in video. There is a link for a discord group where you can get immediate answers from other members in case you choose one of those ways, docker is the less intrusive with your system (no need to install compilers, etc)

TriniHam, 2022-03-21:
> I have not seen a video of the process, I guess everyone stresses before it works and then forgets after it does so nobody has taken the time to document it in video. There is a link for a discord [...]

Ok thanks, is there a link for this discord group?

ArjanOnwezen, 2022-03-21:
https://discord.gg/tuwVMv3

ArjanOnwezen, 2022-04-01:
It looks like it's not possible due to hardware limitation to change LCD brightness.
Feel free to open the issue, if it is possible.

closed 2022-04-01 by ArjanOnwezen