portapack-mayhem/mayhem-firmware #505: What version is the Hack RF?

TriniHam, 2022-03-16:
Is the Hack RF app at this version ? [release 2021.03.1](https://github.com/greatscottgadgets/hackrf/releases/tag/v2021.03.1) If not is there a particular way to update it? if yes where can I see a guide to do so.

Reason for asking this is, this [hackrf-spectrum-analyzer](https://github.com/pavsa/hackrf-spectrum-analyzer) isn't working with the current Mayhem 1.4.3.

If everything is updated is there a similar tool I can use and please not SDRSharp, much thanks.

voxo22, 2022-03-16:
Yeah, I have asked Pavol (pavsa), creator of the spectrum analyzer to upgrade the code of his software, he is quite busy for now, but he is open to the idea. I have done a lot of improvements of the spectrum analyzer and planned to commit them to the code. Personally there is not big deal to "downgrade" to 1.3.1 for the use of analyzer and then upgrade back to 1.4.3. Its very simple and quick. Or, better to have 2 hackrfs, one with 1.3.1 release for analyzer use, the other one with the latest for portapack features.

TriniHam, 2022-03-17:
> Yeah, I have asked Pavol (pavsa), creator of the spectrum analyzer to upgrade the code of his software, he is quite busy for now, but he is open to the idea. I have done a lot of improvements of the [...]

Ok thanks for the update but yes I understand

ArjanOnwezen, 2022-03-18:
As mentioned there an issue with the spectrum analyzer. This isn't related to this code base. Closing the issue

closed 2022-03-18 by ArjanOnwezen

TriniHam, 2022-03-20:
**UPDATE!!!**   
@ArjanOnwezen , @eried and everyone.

There's a version that works to this date of comment (March / 20th / 2022.) [hackrf_spectrum_analyzer.zip](https://github.com/pavsa/hackrf-spectrum-analyzer/files/7676635/hackrf_spectrum_analyzer.zip) 

What's needed apart from the normal ? just Java JRE 64bit v1.8+:
[Oracle](https://download.oracle.com/otn/java/jdk/8u321-b07/df5ad55fdd604472a86a45a217032c7d/jre-8u321-windows-x64.exe) 
OR 
[Java](
https://java.com/en/download/)
See original post for more info as to what you may need: [here](https://github.com/pavsa/hackrf-spectrum-analyzer#installation)

Don't have down graded or anything like that works on Mayhem 1.4.3
Hope it helps 👍

voxo22, 2022-03-20:
TriniHam: Wow! Yeah, you are right, man!. But the only difference is newer hackrf_sweep.dll in win32-x86-64 folder. Just replace it in old code and everything works! Where did you find this package? It is newer (7.12.2021) than latest code.
Many thanks, dude!

TriniHam, 2022-03-20:
> TriniHam: Wow! Yeah, you are right, man!. But the only difference is newer hackrf_sweep.dll in win32-x86-64 folder. Just replace it in old code and everything works! Where did you find this package? It is newer (7.12.2021) than latest code. Many thanks, dude!

Ok thanks for that extra piece of info and research bro , alot of it😂. 

_Wish I could give credit to where the links came from but too many tabs. I just downloaded  everything I saw and debug._ 

BTW what made you say its newer?

voxo22, 2022-03-21:
> > Ok thanks for that extra piece of info and research bro , alot of it😂. Happy to see it helped someone.
>  _Wish I could give credit to where the links came from but too many tabs. I just downloaded everything I saw and debug._
>  BTW what made you say its newer?
> 
The latest code of hackrf-spectrum-analyzer was added 18-apr-2020 (I have asked Pavol to implement RF amp switch), so the official release has timestamp of apr 2020. The files in the package you have linked has timestamp dec 2021. So it seems weird to me where it came from ;-)

TriniHam, 2022-03-21:
> The latest code of hackrf-spectrum-analyzer was added 18-apr-2020 (I have asked Pavol to implement RF amp switch), so the official release has timestamp of apr 2020. The files in the package you [...]

Ohh lol. it could have possible been a forked