portapack-mayhem/mayhem-firmware #507: My "feature review"

voxo22, 2022-03-21:
Hi guys,
first many many thanks for amazing piece of firmware. I am daily user of hackrf incl. portapack and now I own 4 pcs of them. Great tool for learning, simulating, browsing radiowaves.
During use of portapack I realized that many features are very cool now, but some would be cool, if they were completed, other miss at all.
I have noted them into a small file and I would like to know, if there is any chance to be completed, or written from scratch. Also, can you please mark those, which never can run on portapack due to CPU/memory,etc problems in the future? Thank you!
73

Rx buggy/incompleted: 
• AFSK: seems can succesfully decode TEDI/LCR, but otherwise does not work?
• Audio: SPEC always displays red line in the middle of waterfall
• Audio: WFM limited use of signal spectrum above waterfall
• Analog TV: completely useless. Even while very good signal, just very crapy bw picture w/o sound
• ERT: very limited use, app writes to log, but nothing shows on screen
• Radiosonde: limited types of sondes
• TPMS cars: limited types of sensors, app writes to log, but nothing shows on screen

Tx buggy/incompleted:
• BHT Xy/EP: very limited use
• Key FOB: useless for this time
• LGE Tool: useless for this time
• BurgerPgr: useless for this time
• TEDI/LCR: generally more info needed (freqs, modem setup, etc) - message can be decoded by AFSK Rx
• RDS: needs to be completed

Missing features/Nice to have:
• Simple spectrum analyzer (large graph Freq2Amplitude) with tiny (switchable) WF at the bottom (not critical), variable span, peak display with switchable AVG/MAX/MAX HOLD colored lines
• Audio/AM: variable bandwidth, digital broadcast DRM30/DRM+ decoding
• Audio/NFM: DSD (simple digital DMR/dPMR/NXDN/D-Star/C4FM decoder (for unencrypted channels))
• Audio/WFM: FM stereo + RDS (is Phone jack/audio IC Stereo ready?)
• Audio/RAW I/Q (stereo) samples output to jack
• Audio/CW Morse variable speed decoder
• Audio/RTTY decoder
• Audio/DAB Rx
• "rtl_433 by Merbanan" decoding options for multiple protocols, option for freq change
• Soundbrd: modulation type (FM/AM/SSB) option
• Microphone app: modulation type (FM/AM/SSB) option for Rx section
• Options/Signal generator: modulation type (FM/AM/SSB) option
• Wave player for listening of recorded audio (w/ stereo)
• DATV Rx
• SSTV Rx
• SigFox/LoRa/ZigBee/etc Rx/decoder
• AFSK decoder
• Tetra decoder
• AIS Tx
• Zigbee/LoRa/LoWPAN Tx
• Universal text file viewer (or editor)
• Map zooming level in options

ArjanOnwezen, 2022-03-21:
Hi @voxo22. Great to see your enthusiasm.
It might better though to split requests into separate cards, after checking which cards are not already present. 
This way it might motivate others to develop and/or comment on individual items.

voxo22, 2022-03-23:
Sure, I get it. But I didn't want to clutter up the Issues section with dozens of new tickets. Rather, it would help to have an overall view of the list, at least which of those features you think will never run on portapack.

ArjanOnwezen, 2022-03-26:
@voxo22 Adding dozens of tickets doesn't harm. Just make sure you don't add duplicates. Furthermore you can also check already closed tickets, since it might include comments about hardware limitations for example.
Also if you like learning, don't be afraid to add and/or complete apps yourself.
If will close this ticket, since it includes too much to be able to work on.

closed 2022-03-26 by ArjanOnwezen