portapack-mayhem/mayhem-firmware #509: TPMS Transmitter

jLynx, 2022-03-23:
We currently have a TPMS Receiver, but it would be awesome to also be able to transmit TPMS signals.

I originally suggested this a few years ago on the Havoc firmware, but since that has been abandoned, I thought I would move it over to here https://github.com/furrtek/portapack-havoc/issues/254

jLynx, 2022-03-24:
Currently working on this here https://github.com/jLynx/portapack-mayhem/tree/PSI

jLynx, 2022-03-24:
Some more info here https://www.s4ur0n.com/assets/media/hc0n2k.pdf

jLynx, 2022-03-24:
https://github.com/JoeSc/Subaru-TPMS-Spoofing

janicektrnecka, 2022-04-30:
I would love to be able to spoof such signal, just be reading the serial and spoofing the value for specific serial# of TPMS.  Having this option directly on portapack without messing with computer would be nice!

jLynx, 2023-02-06:
No updates with this yet @simkard69