portapack-mayhem/mayhem-firmware #511: Add support for PSI in TPMS

jLynx, 2022-03-24:
It would be great if we could switch between kPa and PSI

jLynx, 2022-03-24:
Working on this here https://github.com/jLynx/portapack-mayhem/tree/PSI

jLynx, 2022-03-24:
PR's are here https://github.com/eried/portapack-mayhem/pull/512 and here https://github.com/eried/portapack-mayhem/pull/513

eried, 2022-03-24:
which implementation you prefer?

jLynx, 2022-03-24:
Honestly I dont know. For v2 I changed RecentEntriesColumns which is a fundamental component to get it to work (Should be pretty safe. Everything I tested still works fine)

But with v1 I didnt have to do that, but I feel like that creates a bunch of code duplication (although not true duplication).

So I am 50/50 with either or. But if I **had** to choose, I would probably say v2 (https://github.com/eried/portapack-mayhem/pull/513)

eried, 2022-03-24:
I am just more prone to go for the other because is less modifications only :P

jLynx, 2022-03-24:
Yeah that's fine. I am happy with either

jLynx, 2022-03-28:
Any update with getting version 1 merged @eried?

jLynx, 2022-03-28:
Closed as https://github.com/eried/portapack-mayhem/pull/512 was merged in

closed 2022-03-28 by jLynx